platformio/platformio-home #1365: Could not initialize project

orrmany, 2020-09-25:
As a preparation to create an issue for https://community.platformio.org/t/mysterious-build-failure-after-platformio-update/15986 I tried to create a new project in PlatformIO IDE. The used PlatformIO Core is 5.0.1, Home 3.0.0.
The used VSCode is  Version: 1.49.1 (user setup)
Commit: 58bb7b2331731bf72587010e943852e13e6fd3cf
Date: 2020-09-16T23:27:51.792Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.17763

I have selected board as Adafruit Feather nRF52840 Express, framework as Arduino. 

The creation of the empty project has failed:

PIO Core Call Error: "The current working directory C:\\Users\\egbozie\\OneDrive - Ericsson AB\\Documents\\PlatformIO\\Projects\\Core5Test will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\egbozie\\OneDrive - Ericsson AB\\Documents\\PlatformIO\\Projects\\Core5Test\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Could not install package 'framework-arduinoadafruitnrf52 @ ~1.1900.0' for 'windows_amd64' system"

manuelbl, 2020-09-30:
There are known issues with both OneDrive directories and path with spaces or non-ASCII letters. Can you try in a different directory?

orrmany, 2020-09-30:
Sure, I will, but it has worked prior mid-August, i.e., prior the 5.x core...

orrmany, 2020-09-30:
ok, restarting both he PlatformIO IDE and the OneDrive agent solved the problem. Thanks @manuelbl

closed 2020-09-30 by orrmany