platformio/platformio-home #1369: Libraries: Updates

wefleenor, 2020-09-25:
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n  File \"C:\\Users\\Bill_\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"C:\\Users\\Bill_\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"C:\\Users\\Bill_\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n    rv = self.invoke(ctx)\r\n  File \"C:\\Users\\Bill_\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"C:\\Users\\Bill_\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\Bill_\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\Bill_\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"C:\\Users\\Bill_\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"C:\\Users\\Bill_\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\r\n    return f(get_current_context(), *args, **kwargs)\r\n  File \"C:\\Users\\Bill_\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib\\command.py\", line 247, in lib_update\r\n    lm = LibraryPackageManager(storage_dir)\r\n  File \"C:\\Users\\Bill_\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\library.py\", line 30, in __init__\r\n    PackageType.LIBRARY, package_dir or get_project_global_lib_dir()\r\n  File \"C:\\Users\\Bill_\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\base.py\", line 54, in __init__\r\n    self.package_dir = self.ensure_dir_exists(package_dir)\r\n  File \"C:\\Users\\Bill_\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\base.py\", line 98, in ensure_dir_exists\r\n    assert os.path.isdir(path)\r\nAssertionError\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

DAKarp, 2020-09-27:
See https://github.com/platformio/platformio-home/issues/1309

ivankravets, 2020-10-30:
Please open PlatformIO CLI and type `pio upgrade`. Restart VSCode. Is fixed in the latest PlatformIO Core 5.0.2

closed 2020-10-30 by ivankravets