platformio/platformio-home #1397: ER: Bring back the install button (GlobalStorage) for libraries

DAKarp, 2020-09-27:
Enhancement request:

The latest version of PIO took away the ability to quickly add a library to global storage. This means that if you have a library you want to use in a dozen different projects, you have to manually install it into each one. And I find myself in precisely that situation, which is a pity because PlatformIO has been such a good tool up until now.

Before anyone jumps on this with "well, that's how arduino ide does it" or "global storage has this problem and that problem," I do understand the downsides of using GlobalStorage for libraries, particularly for folks who don't know what they're doing. And I do see the benefit of limiting the scope of an installation to a particular project, and moreso making that the default. But please don't make it the *only* option.

And yes, I do know you can install it manually with CLI but that's a total nuisance and a real waste of my time considering (A) sometimes I quickly need to test several libraries before finding one that meets my needs, (B) the CLI is extremely buried in the VSCode UI, and (C) it used to be -- and could still be -- so easy.

So how about this? Add an option to the PlatformIO IDE configuration so show a "Global Install" button next to the default-but-much-less-useful "add to project" button along with a stern warning to scare away people who don't understand what any of this means. Please?

dee-max, 2020-09-28:
Yes its a drag. Also the libraries that i am trying to install is not working. Please could you have a look at the issue.

https://github.com/platformio/platformio-home/issues/1400

Thanks in advance.

DAKarp, 2020-10-20:
So are there plans to fix this? As it is, the Libraries -> Updates UI is totally busted. Would love to see this resolved as I use PIO with vscode every day.

trizmark, 2020-10-21:
I have questioned the removal of this feature as well.

Having thought about it, the global storage is not the best way to deal with libraries. Defining per-project dependencies is the way forward. This way, when my teammates build my code, I can be sure that there will be no issues due to a mismatch in library version.

Have a look at the lib_deps flag in platformio.ini. It works like a charm.

Here's an example from one of my projects. By specifying the library dependencies like this, I can be sure that the build will be successful on any machine without having to rely on libraries in global storage.

lib_deps=
  knolleary/PubSubClient @ ^2.8
  tzapu/WifiManager @ ^0.15.0
  adafruit/Adafruit NeoPixel @ ^1.6.0
  paulstoffregen/Time @ ^1.6

dee-max, 2020-10-21:
yes i can understand that but not all people need 'the best way'. After the update for me everything thing else broke; This is not a very well tested platform (which i understand as its free) but due to this change lot of people are facing issue. After the update am getting errors & nothing is working; i had to stop using it now.
At least people should have the **option** to keep the existing way or use a private set of libraries for each project.  Our code that we have been working on for months/years have stopped compiling now just because _somebody thinks its better a certain way_. Well some of us do not; we want to keep it simple.

trizmark, 2020-10-21:
You can still install libs to global storage using CLI.

E.g:
pio lib -g install "adafruit/Adafruit NeoPixel"

dee-max, 2020-10-21:
thank you - i will try it out

DAKarp, 2020-10-21:
> Having thought about it, the global storage is not the best way to deal with libraries. Defining per-project dependencies is the way forward. This way, when my teammates build my code, I can be sure that there will be no issues due to a mismatch in library version.

I've thought about it, too.

Just yesterday, Adafruit released a minor point release to the Adafruit GFX Library. I use this library in at least a dozen different projects. I definitely want this update in all of them, but I don't want to have to update a dozen different ini files. This is, for me, the most common use case by far.

But it also happens that a library update breaks things, either due to sloppy engineering or an architecture change. Either way, this is annoying, but much more the exception than the rule. Of course, when this happens, it's easy enough to roll back to an earlier version of the library if needed. In the very narrow use case of different projects requiring different versions of a library, I'd certainly prefer per-project dependencies. But that's very rare.

So, yes, it's true that I can use the nice library UI to search for a library and then open up the CLI and type this tedious command:

`pio lib -g install "Adafruit GFX Library"`

every single time I want to install a library.

And it's true that I can update my libraries by again opening up the CLI and typing:

`pio update`

But isn't the point of a UI to make things easier and not harder? I'd rather be spending my time coding than mucking around in the CLI because of a hobbled library manager.

closed 2020-12-27 by stale[bot]

sambartle, 2021-01-03:
I also find this to be an annoying removal.. It must be adding hours to my week every week messing with libs all over multiple projects.. which used to all be managed through the GUI global option just fine. Id also like to see it returned. It doesn't need to replace the current method for people who prefer it.

I know I can do it through the command line.. but if I wanted to be doing that I wouldn't be using the VS Code GUIO addon..