platformio/platformio-home #1449: Could not load platform updates

PGlyre, 2020-10-05:
PIO Core Call Error: "Error: Traceback (most recent call last):\n  File \"/Users/patrick/.platformio/penv/lib/python3.8/site-packages/platformio/__main__.py\", line 109, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"/Users/patrick/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 829, in __call__\n    return self.main(*args, **kwargs)\n  File \"/Users/patrick/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 782, in main\n    rv = self.invoke(ctx)\n  File \"/Users/patrick/.platformio/penv/lib/python3.8/site-packages/platformio/commands/__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"/Users/patrick/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/patrick/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/patrick/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 1066, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"/Users/patrick/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 610, in invoke\n    return callback(*args, **kwargs)\n  File \"/Users/patrick/.platformio/penv/lib/python3.8/site-packages/platformio/commands/platform.py\", line 393, in platform_update\n    and not PlatformFactory.new(pkg).are_outdated_packages()\n  File \"/Users/patrick/.platformio/penv/lib/python3.8/site-packages/platformio/platform/_packages.py\", line 133, in are_outdated_packages\n    if self.pm.outdated(\n  File \"/Users/patrick/.platformio/penv/lib/python3.8/site-packages/platformio/package/manager/_update.py\", line 48, in outdated\n    self.pick_best_registry_version(reg_pkg[\"versions\"]) or {}\nKeyError: 'versions'\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

dianlight, 2020-10-07:
Mee too:

PIO Core Call Error: "Error: Traceback (most recent call last):\n  File \"/Users/ltarantino/.platformio/penv/lib/python3.8/site-packages/platformio/__main__.py\", line 109, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"/Users/ltarantino/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 829, in __call__\n    return self.main(*args, **kwargs)\n  File \"/Users/ltarantino/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 782, in main\n    rv = self.invoke(ctx)\n  File \"/Users/ltarantino/.platformio/penv/lib/python3.8/site-packages/platformio/commands/__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"/Users/ltarantino/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/ltarantino/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/ltarantino/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 1066, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"/Users/ltarantino/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 610, in invoke\n    return callback(*args, **kwargs)\n  File \"/Users/ltarantino/.platformio/penv/lib/python3.8/site-packages/platformio/commands/platform.py\", line 393, in platform_update\n    and not PlatformFactory.new(pkg).are_outdated_packages()\n  File \"/Users/ltarantino/.platformio/penv/lib/python3.8/site-packages/platformio/platform/_packages.py\", line 133, in are_outdated_packages\n    if self.pm.outdated(\n  File \"/Users/ltarantino/.platformio/penv/lib/python3.8/site-packages/platformio/package/manager/_update.py\", line 48, in outdated\n    self.pick_best_registry_version(reg_pkg[\"versions\"]) or {}\nKeyError: 'versions'\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

ivankravets, 2020-10-30:
Please open PlatformIO CLI and type `pio upgrade`. Restart VSCode. Is fixed in the latest PlatformIO Core 5.0.2

closed 2020-10-30 by ivankravets