platformio/platformio-home #1471: Could not initialize project

cirobruno, 2020-10-11:
Already one year trying to start the first project with VS Code. Till now, just error messages...
It looks like one should be prepared to spend more time with the IDE configuration than coding for the device itself.
I can't see any advantage so far.
Better keep on Arduino IDE, isn't it? Or maybe trying Arduino Pro instead.

PIO Core Call Error: "The current working directory /Users/cirobruno/Documents/PlatformIO/ESP32 Teste 03 will be used for the project.\n\nThe next files/directories have been created in /Users/cirobruno/Documents/PlatformIO/ESP32 Teste 03\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html\nPLATFORM: Espressif 32 (2.0.0) > DOIT ESP32 DEVKIT V1\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\nPACKAGES: \n - framework-espidf 3.40100.200827 (4.1.0) \n - tool-cmake 3.16.4 \n - tool-esptoolpy 1.20600.0 (2.6.0) \n - tool-ninja 1.9.0 \n - toolchain-esp32ulp 1.22851.191205 (2.28.51) \n - toolchain-xtensa32 2.80200.200827 (8.2.0)\nError: Detected a whitespace character in project paths.\n========================== [FAILED] Took 0.68 seconds =========================="

cirobruno, 2020-10-11:
I've been looking at the list of recently posted issues. Absolute majority consist of "Could not".
After all, what does it really "CAN"?

ivankravets, 2020-10-30:
See https://docs.platformio.org/en/latest/frameworks/espidf.html#limitations

closed 2020-10-30 by ivankravets