platformio/platformio-home #1520: Could not initialize project

michael-godoi, 2020-10-18:
PIO Core Call Error: "The current working directory C:\\Users\\Michael\\hello_word will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\Michael\\hello_word\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html\r\nPLATFORM: Espressif 32 (2.0.0) > DOIT ESP32 DEVKIT V1\r\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\r\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\r\nPACKAGES: \r\n - framework-espidf 3.40100.200827 (4.1.0) \r\n - tool-cmake 3.16.4 \r\n - tool-esptoolpy 1.20600.0 (2.6.0) \r\n - tool-idf 1.0.1 \r\n - tool-mconf 1.4060000.20190628 (406.0.0) \r\n - tool-ninja 1.9.0 \r\n - toolchain-esp32ulp 1.22851.191205 (2.28.51) \r\n - toolchain-xtensa32 2.80200.200827 (8.2.0)\r\nInstalling ESP-IDF's Python dependencies\r\n'C:\\Users\\Michael' nÆo ‚ reconhecido como um comando interno\r\nou externo, um programa oper vel ou um arquivo em lotes.\r\n*** Error 1\r\nInstalling windows-curses package\r\n'C:\\Users\\Michael' nÆo ‚ reconhecido como um comando interno\r\nou externo, um programa oper vel ou um arquivo em lotes.\r\n*** Error 1\r\nError: Detected a whitespace character in project paths.\r\n========================== [FAILED] Took 1.39 seconds =========================="

michael-godoi, 2020-10-18:
Se eu utilizar a placa do Arduino eu consigo criar o projeto, porém se eu utilizar o DOIT não consigo de maneira alguma.

ivankravets, 2020-10-30:
See https://docs.platformio.org/en/latest/frameworks/espidf.html#limitations

closed 2020-10-30 by ivankravets