platformio/platformio-home #1566: Unable to update "platformio/contrib-pysite"

RobThree, 2020-10-25:
When updating PlatformIO I get:

    platformio update
    Updating platformio/contrib-piohome           3.3.0 @ ~3.3.0                     [Up-to-date]
    Updating platformio/contrib-pysite            2.38.191020 @ ~2.38.0              [Outdated 2.38.201019]
    Tool Manager: Installing platformio/contrib-pysite @ 2.38.201019
    Unpacking  [####################################]  100%
    Tool Manager: contrib-pysite @ 2.38.201019 has been installed!
    Tool Manager: Removing contrib-pysite @ 2.38.191020
    Tool Manager: contrib-pysite @ 2.38.191020 has been removed!
    Updating platformio/tool-unity                1.20500.200612 @ ~1.20500.0        [Up-to-date]

And when retrying:

    platformio update
    Updating platformio/contrib-piohome           3.3.0 @ ~3.3.0                     [Up-to-date]
    Updating platformio/contrib-pysite            2.38.201019 @ ~2.38.0              [Incompatible 2.39.201019]
    Updating platformio/tool-unity                1.20500.200612 @ ~1.20500.0        [Up-to-date]

[image]

Unfortunately I have no clue how to fix this.

cyba3r, 2020-11-19:
Same issue here under Windows

RobThree, 2020-11-19:
[image]

Yep, getting the same under windows.

yuanyanhui, 2020-11-22:
Same issue in Windows. Also cannot debug. In the debug console, see a lot of errors from python programs.

yuanyanhui, 2020-11-26:
According to this post, the incompatible issue is not a problem.
https://community.platformio.org/t/platformio-libraries-updates-not-working-after-updating-to-5-x/16638/13

ivankravets, 2021-01-20:
The latest PlatformIO Core 5.0.5-dev does not require pysite package. We migrated PIO Home to native Python 3.

closed 2021-01-20 by ivankravets