platformio/platformio-home #1583: Library inspection throws utf-8 decoding error

maxgerhardt, 2020-10-27:
With the new PIO dev core installed, library update buttons are mostly not greyed out anymore, but for some library depdencies, a fatal error is thrown.

When manually starting `pio home` and navigating to the Libraries->Update section

[image]

```
Open PlatformIO Home in your browser by this URL => http://127.0.0.1:8008
PIO Home has been started. Press Ctrl+C to shutdown.
Error: {'code': 4003, 'message': 'PIO Core Call Error', 'data': "'utf-8' codec can't decode byte 0xf6 in position 204: invalid start byte"}
```

This environment has `.libdeps` folder which contains the `ustd` library with the following `library.json`

```
{
    "name": "ustd",
    "keywords": "std, queue, map, array",
    "description": "Small, highly portable library for queues, maps and arrays",
    "repository": {
        "type": "git",
        "url": "https://github.com/muwerk/ustd.git"
    },
    "authors": [{
        "name": "Dominik Schlösser",
        "email": "dsc@dosc.net",
        "maintainer": true
    }, {
        "name": "Leo Moll",
        "email": "leo@yeasoft.com",
        "maintainer": true
    }],
    "version": "0.2.1",
    "frameworks": "*",
    "platforms": "*"
}
```

```
>pio system info
--------------------------  ---------------------------------------------------------------------------
PlatformIO Core             5.0.2b3
Python                      3.7.6-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\Maxi\.platformio
PlatformIO Core Executable  C:\Users\Maxi\AppData\Local\Programs\Python\Python37\Scripts\platformio.exe
Python Executable           c:\users\maxi\appdata\local\programs\python\python37\python.exe
Global Libraries            4
Development Platforms       19
Tools & Toolchains          102
--------------------------  ---------------------------------------------------------------------------
```

ivankravets, 2020-10-27:
Thanks for the report! Fixed in the latest PIO Core Dev. Please re-run `pio upgrade --dev`.

closed 2020-10-27 by ivankravets

maxgerhardt, 2020-10-27:
Confirmed solved, no errors are thrown anymore and the update button shows up.

[image]

ivankravets, 2020-10-27:
Need to fix a few issues and we can have the official 5.0.2 release :) https://github.com/platformio/platformio-core/milestone/83