platformio/platformio-home #1591: Libraries: Could not install library

freezeghost, 2020-10-28:
PIO Core Call Error: "Library Storage: D:\\!Prog\\Arduino\\My-TTGO-Watch-master\\.pio\\libdeps\\ttgo-t-watch\r\nLibrary Manager: Installing xinyuan-lilygo/TTGO TWatch Library @ ^1.3.0\r\nUnpacking...\r\nLibrary Manager: Warning! Package Mirror: CRC check failed 0xed05a807 != 0x23a65ce7\r\nLibrary Manager: Looking for another mirror...\r\nUnpacking...\r\nLibrary Manager: Warning! Package Mirror: CRC check failed 0xed05a807 != 0x23a65ce7\r\nLibrary Manager: Looking for another mirror...\r\nUnpacking...\r\nLibrary Manager: Warning! Package Mirror: CRC check failed 0xed05a807 != 0x23a65ce7\r\nLibrary Manager: Looking for another mirror...\r\n\n\nError: Could not install package 'xinyuan-lilygo/TTGO TWatch Library @ ^1.3.0' for 'windows_amd64' system"

jurov, 2020-11-03:
Confirm. But it just fixed itself somehow.

ivankravets, 2020-11-03:
Sorry, fixed in the registry. Please re-run `pio update.`

closed 2020-11-03 by ivankravets

freezeghost, 2020-11-03:
great it is work now. Thank yoy !