platformio/platformio-home #1598: Libraries: Updates (complete library management)

Schawen, 2020-10-30:
[image]

Error message:

` PIO Core Call Error: "b''\n\nError: Traceback (most recent call last):\r\n  File \"c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n    rv = self.invoke(ctx)\r\n  File \"c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\r\n    return f(get_current_context(), *args, **kwargs)\r\n  File \"c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib\\command.py\", line 290, in lib_update\r\n    json_result[storage_dirs[0]] if len(storage_dirs) == 1 else json_result\r\n  File \"c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 97, in <lambda>\r\n    click.echo = lambda *args, **kwargs: _safe_echo(0, *args, **kwargs)\r\n  File \"c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 91, in _safe_echo\r\n    click_echo_origin[origin](*args, **kwargs)\r\n  File \"c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\click\\utils.py\", line 272, in echo\r\n    file.write(message)\r\n  File \"C:\\Users\\MiKo\\.platformio\\python3\\lib\\encodings\\cp1252.py\", line 19, in encode\r\n    return codecs.charmap_encode(input,self.errors,encoding_table)[0]\r\nUnicodeEncodeError: 'charmap' codec can't encode character '\\u2714' in position 6997: character maps to <undefined>\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"`

(formatted:)

`PIO Core Call Error: "b''
Error: Traceback (most recent call last):
File "c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main
cli() # pylint: disable=no-value-for-parameter
File "c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__
return self.main(*args, **kwargs)
File "c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main
rv = self.invoke(ctx)
File "c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke
return super(PlatformioCLI, self).invoke(ctx)
File "c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke
return ctx.invoke(self.callback, **ctx.params)
File "c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke
return callback(*args, **kwargs)
File "c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func
return f(get_current_context(), *args, **kwargs)
File "c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib\\command.py\", line 290, in lib_update
json_result[storage_dirs[0]] if len(storage_dirs) == 1 else json_result
File "c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 97, in <lambda>
click.echo = lambda *args, **kwargs: _safe_echo(0, *args, **kwargs)
File "c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 91, in _safe_echo
click_echo_origin[origin](*args, **kwargs)
File "c:\\users\\miko\\.platformio\\penv\\lib\\site-packages\\click\\utils.py\", line 272, in echo
file.write(message)
File "C:\\Users\\MiKo\\.platformio\\python3\\lib\\encodings\\cp1252.py\", line 19, in encode
return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\u2714' in position 6997: character maps to <undefined>

============================================================

An unexpected error occurred. Further steps:
* Verify that you have the latest version of PlatformIO using
`pip install -U platformio` command
* Try to find answer in FAQ Troubleshooting section
https://docs.platformio.org/page/faq.html
* Report this problem to the developers
https://github.com/platformio/platformio-core/issues

============================================================`

I've already uninstalled plattformio, deleted the lib directory and reinstalled. 
Upgraded the PIO core to the latest dev version and copied the saved lib folder to the pio directory.
( https://community.platformio.org/t/update-libraries-not-possible/16807/9 )

`pio --version
PlatformIO, version 5.0.2rc1`

[image]

and 

[image]

ivankravets, 2020-10-30:
Could you provide a list of global libraries? We need to know which library causes this issue. You can try to remove lib-by-lib from `%HOMEDIR/.platformio/lib%` and find the problematic library.

Schawen, 2020-10-30:
These are all folders in directory **C:\Users\MiKo\.platformio\lib**

AccelStepper			
Adafruit ADXL343		
Adafruit BME280	Library	
Adafruit BusIO		
Adafruit Circuit Playground	
Adafruit FONA Library	
Adafruit GFX Library	
Adafruit GPS Library	
Adafruit ILI9341		
Adafruit LED Backpack Library
Adafruit MQTT Library	
Adafruit SleepyDog Library	
Adafruit SSD1306		
Adafruit STMPE610		
Adafruit TouchScreen		
Adafruit Unified Sensor	
Adafruit Zero DMA Library
Adafruit Zero FFT Library
Adafruit Zero PDM Library
ArduinoJson			
DallasTemperature			
DHT sensor library	
DS3232RTC			
ESPAsyncUDP			
ILI9488_ID4667			
LiquidCrystal_ID887			
Mouse			
NtpClientLib_ID727			
OneWire_ID1			
PubSubClient			
RTC			
RTClib			
SD			
TFT_eSPI			
Time_ID44			
WaveHC			
WebSockets			
WifiManager_ID567			
XPT2046_Touchscreen			

**I tried to search for "u+2714" using powershell, but I'm not sure whether I can trust the result or not ...**

`PS C:\Users\MiKo> sls

Cmdlet Select-String an der Befehlspipelineposition 1
Geben Sie Werte für die folgenden Parameter an:
Pattern[0]: ✔
Pattern[1]:
Path[0]: C:\Users\MiKo\.platformio\lib\*.*
Path[1]:
PS C:\Users\MiKo>`

Do you have any other idea on how to search the directory for u+2714?

maxgerhardt, 2020-10-30:
Yeah the description contains these Unicode chars:

https://github.com/bblanchon/ArduinoJson/blob/6.x/library.json#L4

```
  "description": "A simple and efficient JSON library for embedded C++. ArduinoJson supports ✔ serialization, ✔ deserialization, ✔ MessagePack, ✔ fixed allocation, ✔ zero-copy, ✔ streams, ✔ filtering, and more. It is the most popular Arduino library on GitHub ❤❤❤❤❤. Check out arduinojson.org for a comprehensive documentation.",
```

There's the checkmark we've been looking for.

ivankravets, 2020-10-30:
RC2 is out. Please open PlatformIO CLI and type `pio upgrade --dev`. Restart VSCode. Does it work now?

maxgerhardt, 2020-10-30:
[image]

Works for me now

ivankravets, 2020-10-30:
[PlaformIO Core 5.0.2 is out!](https://community.platformio.org/t/platformio-core-5-0/15721/8?u=ivankravets)

THANKS FOR THE REPORTS!!! 😊

closed 2020-10-30 by ivankravets

Schawen, 2020-10-31:
Sorry, doesn't work for me.

Updated and restarted PIO:

`PlatformIO, version 5.0.2`

but

[image]

Copied ArduinoJson lib to the lib dir  and tested:

[image]

Schawen, 2020-10-31:
Also tested with same

`PS D:\Dokumente\PlatformIO\Projects\THP\THP-Sender_089> pio lib -g install ArduinoJson@6.16.0
Please wait while upgrading PlatformIO...
PlatformIO has been successfully upgraded to 5.0.2!

*****************************************************************************************************************************************************************If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
*****************************************************************************************************************************************************************
Library Storage: C:\Users\MiKo\.platformio\lib
Library Manager: Installing ArduinoJson @ 6.16.0
Library Manager: ArduinoJson @ 6.16.0 has been installed!`

ivankravets, 2020-10-31:
Could you restart your machine?

Schawen, 2020-10-31:
Same situation after restart - same error.

maxgerhardt, 2020-10-31:
Still works for me and I'm on `pio --version` == 5.0.2. 

[image]

What "Core" version is being shown in the PIO home start page?

Schawen, 2020-10-31:
[image]

ivankravets, 2020-10-31:
It’s not the latest stable version. It seems you have multiple PlatformIO Cores in a system. Please remove duplicates https://docs.platformio.org/en/latest/faq.html

ivankravets, 2020-10-31:
Try to open PlatformIO CLI and type `pio upgrade`. Restart VSCode.

Schawen, 2020-10-31:
This was already active:

[image]

Uninstalled PIO from system cli:
`
Microsoft Windows [Version 10.0.19042.572]
(c) 2019 Microsoft Corporation. Alle Rechte vorbehalten.

C:\WINDOWS\system32>pip uninstall platformio
Found existing installation: platformio 5.0.2
Uninstalling platformio-5.0.2:
  Would remove:
    c:\users\miko\appdata\local\programs\python\python38\lib\site-packages\platformio-5.0.2-py3.8.egg-info
    c:\users\miko\appdata\local\programs\python\python38\lib\site-packages\platformio\*
    c:\users\miko\appdata\local\programs\python\python38\lib\site-packages\scripts\99-platformio-udev.rules
    c:\users\miko\appdata\local\programs\python\python38\lib\site-packages\scripts\docspregen.py
    c:\users\miko\appdata\local\programs\python\python38\lib\site-packages\scripts\fixsymlink.py
    c:\users\miko\appdata\local\programs\python\python38\lib\site-packages\scripts\get-platformio.py
    c:\users\miko\appdata\local\programs\python\python38\lib\site-packages\scripts\install_devplatforms.py
    c:\users\miko\appdata\local\programs\python\python38\scripts\pio-script.py
    c:\users\miko\appdata\local\programs\python\python38\scripts\pio.exe
    c:\users\miko\appdata\local\programs\python\python38\scripts\piodebuggdb-script.py
    c:\users\miko\appdata\local\programs\python\python38\scripts\piodebuggdb.exe
    c:\users\miko\appdata\local\programs\python\python38\scripts\platformio-script.py
    c:\users\miko\appdata\local\programs\python\python38\scripts\platformio.exe
Proceed (y/n)? Y
  Successfully uninstalled platformio-5.0.2
`

Upgrade failed, because I  uninstalled PIO from system?

[image]

But where can I enter the PIO commands now?

ivankravets, 2020-10-31:
1. Remove all Python from the system
2. PlatformIO IDE comes with pre-built Python
3. Remove `%USERHOME%/.platformio/penv` folder and restart VSCode.

P.S: PlatformIO CLI - click on the "terminal" icon on the bottom blue bar or Left side bar > PlatformIO > Quick Access > PlatformIO CLI.

Schawen, 2020-10-31:
`PS D:\Dokumente\PlatformIO\Projects\THP\THP-Sender_089> platformio upgrade

Please wait while upgrading PlatformIO ...
PlatformIO has been successfully upgraded to 5.0.2
Release notes: https://docs.platformio.org/en/latest/history.html`

YEAH - Now it works!!!  (Without the steps from your last post)

Thank you very much for all your time and work!

ivankravets, 2020-10-31:
Great!

P.S: You have multiple Pythons, multiple PlatformIO Cores... It means that problems will love you.

Schawen, 2020-10-31:
I thought that I have uninstalled the duplicate PIO core?

`Found existing installation: platformio 5.0.2
Uninstalling platformio-5.0.2: ...`

And I need Python sometimes for Raspberry programming ...
What should I do?

ivankravets, 2020-10-31:
Ah... now everything is ok! 

Happy coding with PlatformIO! :)

Schawen, 2020-10-31:
Belongs this to the error that loves me? 
I tried to compile a project that previously worked. 
But now I get this error: **fatal error: avr/interrupt.h: No such file or directory**

Processing d1_mini_pro (platform: espressif8266; board: d1_mini_pro; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini_pro.html
PLATFORM: Espressif 8266 (2.6.2) > WeMos D1 mini Pro
HARDWARE: ESP8266 80MHz, 80KB RAM, 16MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 68 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit Unified Sensor> 1.1.4
|-- <PCF8574_ESP> 1.0.10
|   |-- <TinyWireM> 1.1.0
|   |-- <Wire> 1.0
|-- <LiquidCrystal_I2C> 1.1.4
|   |-- <Wire> 1.0
|-- <NTPClient> 3.1.0
|-- <Adafruit BME280 Library> 2.1.1
|   |-- <Adafruit Unified Sensor> 1.1.4
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <DallasTemperature> 3.9.1
|   |-- <OneWire> 2.3.5
|-- <DHT sensor library> 1.4.0
|   |-- <Adafruit Unified Sensor> 1.1.4
|-- <EEPROM> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <NtpClientLib> 3.0.2-beta
|   |-- <Time> 1.6
|   |-- <ESPAsyncUDP> 0.0.0-alpha+sha.697c75a025
|   |-- <Ethernet(esp8266)> 1.0.4
|   |   |-- <SPI> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <Ticker> 1.0
|-- <OneWire> 2.3.5
|-- <PubSubClient> 2.8
|-- <Time> 1.6
|-- <WebSockets> 2.3.1
|   |-- <ESP8266WiFi> 1.0
|   |-- <Ethernet(esp8266)> 1.0.4
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <Hash> 1.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\d1_mini_pro\libd9e\TinyWireM\USI_TWI_Master.cpp.o
C:\users\miko\.platformio\lib\TinyWireM\USI_TWI_Master.cpp:28:27: fatal error: avr/interrupt.h: No such file or directory
 **#include <avr/interrupt.h>**
                           ^
compilation terminated.
Archiving .pio\build\d1_mini_pro\lib256\libNTPClient.a
Archiving .pio\build\d1_mini_pro\libd59\libSPI.a
Compiling .pio\build\d1_mini_pro\libdb0\Adafruit BME280 Library\Adafruit_BME280.cpp.o
Compiling .pio\build\d1_mini_pro\lib656\ESP8266WiFi\BearSSLHelpers.cpp.o
Compiling .pio\build\d1_mini_pro\lib656\ESP8266WiFi\CertStoreBearSSL.cpp.o
Compiling .pio\build\d1_mini_pro\lib656\ESP8266WiFi\ESP8266WiFi.cpp.o
Compiling .pio\build\d1_mini_pro\lib656\ESP8266WiFi\ESP8266WiFiAP.cpp.o
*** [.pio\build\d1_mini_pro\libd9e\TinyWireM\USI_TWI_Master.cpp.o] Error 1
================================================================== [FAILED] Took 5.56 seconds ==================================================================
The terminal process "C:\Users\MiKo\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Added this to platformio.ini:
`
; Checks for the compatibility with frameworks and dev/platforms
lib_compat_mode = strict
`

CODE:
`#include <Arduino.h>

extern "C" { 
  #include <user_interface.h>
}

#include  <TimeLib.h>

#include  <ESP8266WiFi.h>

#include  <ESP8266WebServer.h>

#include  <WebSocketsServer.h>

#include  <WiFiUdp.h>

#include  <ArduinoOTA.h>

#include  <PubSubClient.h>

#include  <Hash.h>

#include  <ESP8266mDNS.h>

#include  <Wire.h>

#include  <OneWire.h>                 //  *DALLAS*

#include  <DallasTemperature.h>

#include  <Adafruit_Sensor.h>

#include  <DHT.h>

#include  <EEPROM.h>

#include  <pcf8574_esp.h>

#include  "LiquidCrystal_I2C.h"

#include  <NtpClientLib.h>

#include  <Adafruit_BME280.h>

#include  "classes.h"

#include  "varTypen.h"

#include  "progmemVars.h"`

Schawen, 2020-10-31:
ok, fixed it with

`lib_ignore = TinyWireM`

in platformio.ini