platformio/platformio-home #1605: Libraries: Could not install library

AntoineLaporte, 2020-10-31:
PIO Core Call Error: "Library Storage: D:\\Documents\\PlatformIO\\Projects\\T-Watch 2020 test\\.pio\\libdeps\\esp32dev\r\nLibrary Manager: Installing xinyuan-lilygo/TTGO TWatch Library @ ^1.3.0\r\nUnpacking...\r\nLibrary Manager: Warning! Package Mirror: CRC check failed 0xed05a807 != 0x23a65ce7\r\nLibrary Manager: Looking for another mirror...\r\nUnpacking...\r\nLibrary Manager: Warning! Package Mirror: CRC check failed 0xed05a807 != 0x23a65ce7\r\nLibrary Manager: Looking for another mirror...\r\nUnpacking...\r\nLibrary Manager: Warning! Package Mirror: CRC check failed 0xed05a807 != 0x23a65ce7\r\nLibrary Manager: Looking for another mirror...\r\n\n\nError: Could not install package 'xinyuan-lilygo/TTGO TWatch Library @ ^1.3.0' for 'windows_amd64' system"

raomin, 2020-11-01:
same here, for the same library...

raomin, 2020-11-01:
I guess this is a pbm with the ttgo library, not with platformio. 
In the meantime you can specify the github repository instead in your platformio.ini file:
```
lib_deps = 
	; TTGO TWatch Library@=1.3.0
	https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library
```

AntoineLaporte, 2020-11-02:
I copied it in local and it works.
I'm a beginner in all this but it was my solution.

ivankravets, 2020-11-03:
Sorry, fixed in the registry. Please re-run `pio update`.

closed 2020-11-03 by ivankravets

Fusseldieb, 2021-01-18:
The "TTGO TWatch Library" is completely broken for me, can't build any project.

Tried specifying versions and even specifying it directly from github.

Always the same errors:

```
Processing ttgo-t-watch (platform: espressif32; board: ttgo-t-watch; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-t-watch.html
PLATFORM: Espressif 32 (2.1.0) > TTGO T-Watch
HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa) 
PACKAGES:
 - framework-arduinoespressif32 3.10004.201016 (1.0.4)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing git+https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library
git version 2.19.2.windows.1
Cloning into 'C:\Users\Valentino\.platformio\.cache\tmp\pkg-installing-z4drovib'...
Library Manager: TTGO TWatch Library @ 1.4.1+sha.d13b203 has been installed!
Found 49 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TTGO TWatch Library> 1.4.1+sha.d13b203
|   |-- <Wire> 1.0.1
|   |-- <ESP32 BLE Arduino> 1.0.1
|   |-- <SPI> 1.0
|   |-- <Ticker> 3.2.0
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <SD(esp32)> 1.0.5
|   |   |-- <FS> 1.0
|   |   |-- <SPI> 1.0
Building in release mode
Compiling .pio\build\ttgo-t-watch\src\main.cpp.o
Generating partitions .pio\build\ttgo-t-watch\partitions.bin
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
Compiling .pio\build\ttgo-t-watch\liba31\Wire\Wire.cpp.o
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
Compiling .pio\build\ttgo-t-watch\lib6b0\BLE\BLE2902.cpp.o
Compiling .pio\build\ttgo-t-watch\lib6b0\BLE\BLE2904.cpp.o
Compiling .pio\build\ttgo-t-watch\lib6b0\BLE\BLEAddress.cpp.o
Compiling .pio\build\ttgo-t-watch\lib6b0\BLE\BLEAdvertisedDevice.cpp.o
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
Compiling .pio\build\ttgo-t-watch\lib6b0\BLE\BLEAdvertising.cpp.o
Compiling .pio\build\ttgo-t-watch\lib6b0\BLE\BLEBeacon.cpp.o
Compiling .pio\build\ttgo-t-watch\lib6b0\BLE\BLECharacteristic.cpp.o
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
Compiling .pio\build\ttgo-t-watch\lib6b0\BLE\BLECharacteristicMap.cpp.o
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
Compiling .pio\build\ttgo-t-watch\lib6b0\BLE\BLEClient.cpp.o
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
Compiling .pio\build\ttgo-t-watch\lib6b0\BLE\BLEDescriptor.cpp.o
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
Compiling .pio\build\ttgo-t-watch\lib6b0\BLE\BLEDescriptorMap.cpp.o
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
Compiling .pio\build\ttgo-t-watch\lib6b0\BLE\BLEDevice.cpp.o
Archiving .pio\build\ttgo-t-watch\liba31\libWire.a
Compiling .pio\build\ttgo-t-watch\lib6b0\BLE\BLEEddystoneTLM.cpp.o
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
Compiling .pio\build\ttgo-t-watch\lib6b0\BLE\BLEEddystoneURL.cpp.o
[... 19 lines cut ...]
C:/Users/Valentino/.platformio/lib/Ticker/Ticker.h:49:7: note: candidate: constexpr Ticker::Ticker(const Ticker&)
 class Ticker {
       ^
C:/Users/Valentino/.platformio/lib/Ticker/Ticker.h:49:7: note:   candidate expects 1 argument, 0 provided
In file included from .pio/libdeps/ttgo-t-watch/TTGO TWatch Library@src-dcd9204adb892bb856cff89f0b8770a2/src/TTGO.h:101:0,
                 from .pio/libdeps/ttgo-t-watch/TTGO TWatch Library@src-dcd9204adb892bb856cff89f0b8770a2/src/LilyGoWatch.h:405,
                 from src\main.cpp:3:
.pio/libdeps/ttgo-t-watch/TTGO TWatch Library@src-dcd9204adb892bb856cff89f0b8770a2/src/drive/tft/bl.h: In member function 'virtual void PWMToneBase::onec(int)':
.pio/libdeps/ttgo-t-watch/TTGO TWatch Library@src-dcd9204adb892bb856cff89f0b8770a2/src/drive/tft/bl.h:109:16: error: 'class Ticker' has no member named 'once_ms'        
         _tick->once_ms<uint8_t>(duration, [](uint8_t channel) {
                ^
.pio/libdeps/ttgo-t-watch/TTGO TWatch Library@src-dcd9204adb892bb856cff89f0b8770a2/src/drive/tft/bl.h:109:31: error: expected primary-expression before '>' token        
         _tick->once_ms<uint8_t>(duration, [](uint8_t channel) {
                               ^
.pio/libdeps/ttgo-t-watch/TTGO TWatch Library@src-dcd9204adb892bb856cff89f0b8770a2/src/drive/tft/bl.h:111:9: warning: left operand of comma operator has no effect [-Wunused-value]
         }, _channel);
         ^
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
*** [.pio\build\ttgo-t-watch\src\main.cpp.o] Error 1
====================================================================== [FAILED] Took 49.69 seconds ======================================================================
The terminal process "C:\Users\Valentino\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

```