platformio/platformio-home #1621: Could not initialize project

hany7nagy, 2020-11-02:
PIO Core Call Error: "DLL load failed: The specified procedure could not be found."

ivankravets, 2020-11-03:
Please open [PlatformIO Core CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and type 
```
pio system info
```

Please share the output here.

hany7nagy, 2020-11-05:
PlatformIO Core             5.0.2
Python                      3.7.7-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1256
PlatformIO Core Directory   C:\Users\hany\.platformio
PlatformIO Core Executable  C:\Users\hany\.platformio\penv\Scripts\platformio.exe
Python Executable           c:\users\hany\.platformio\penv\scripts\python.exe
Global Libraries            0
Development Platforms       2
Tools & Toolchains          12

ivankravets, 2020-12-02:
Duplicate of https://github.com/platformio/platformio-home/issues/1772

closed 2020-12-02 by ivankravets