platformio/platformio-home #1627: Could not initialize project

lamiwe, 2020-11-03:
PIO Core Call Error: "DLL load failed: The specified procedure could not be found."

ivankravets, 2020-11-03:
Please open [PlatformIO Core CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and type 
```
pio system info
```

Please share the output here.

lamiwe, 2020-11-05:
PlatformIO Core             5.0.2
Python                      3.7.7-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\lamiwe\.platformio
PlatformIO Core Executable
 C:\Users\lamiwe\.platformio\penv\Scripts\platformio.exe
Python Executable
c:\users\lamiwe\.platformio\penv\scripts\python.exe
Global Libraries            0
Development Platforms       0
Tools & Toolchains          2

ivankravets, 2020-12-02:
Duplicate of https://github.com/platformio/platformio-home/issues/1772

closed 2020-12-02 by ivankravets