platformio/platformio-home #1630: Could not initialize project

atsteich, 2020-11-03:
PIO Core Call Error: "DLL load failed: Die angegebene Prozedur wurde nicht gefunden."

atsteich, 2020-11-03:
pio system info: 

PlatformIO Core             5.0.2
Python                      3.7.7-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\XX\.platformio
PlatformIO Core Executable  C:\Users\XX\.platformio\penv\Scripts\platformio.exe   
Python Executable           c:\users\XX\.platformio\penv\scripts\python.exe       
Global Libraries            0
Development Platforms       0
Tools & Toolchains          2

ivankravets, 2020-11-04:
Do you use antivirus tools?

atsteich, 2020-11-04:
> Do you use antivirus tools?

Windows defender is aktiv, no antivirus tools..

atsteich, 2020-11-27:
any updates? still not able to create a project, or install a platform

ivankravets, 2020-11-27:
Could you contact me@ikravets.com? I would be happy to provide more info on this issue.

thijsdelft, 2020-11-29:
@ivankravets can you share more info on this issue in this issue? Or provide a link to additional documentation to resolve this issue.
I have the impression mulitple others are facing a similar problem. It would certainly help me ([Issue #1772 ](url))

ivankravets, 2020-12-02:
Duplicate of https://github.com/platformio/platformio-home/issues/1772

closed 2020-12-02 by ivankravets

atsteich, 2020-12-05:
-> The Problem was indeed due to Anaconda... Thanks @ivankravets