platformio/platformio-home #1633: Libraries: Updates

ergerd, 2020-11-04:
PIO Core Call Error: "b''\n\nError: Traceback (most recent call last):\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n    rv = self.invoke(ctx)\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\r\n    return f(get_current_context(), *args, **kwargs)\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib\\command.py\", line 258, in lib_update\r\n    spec = PackageSpec(library)\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\meta.py\", line 119, in __init__\r\n    self._parse(raw)\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\meta.py\", line 222, in _parse\r\n    raw = parser(raw)\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\meta.py\", line 245, in _parse_requirements\r\n    self.requirements = tokens[1].strip()\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\meta.py\", line 162, in requirements\r\n    else semantic_version.SimpleSpec(str(value))\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\semantic_version\\base.py\", line 618, in __init__\r\n    self.clause = self._parse_to_clause(expression)\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\semantic_version\\base.py\", line 1014, in _parse_to_clause\r\n    return cls.Parser.parse(expression)\r\n  File \"c:\\users\\rainerge\\.platformio\\penv\\lib\\site-packages\\semantic_version\\base.py\", line 1034, in parse\r\n    raise ValueError(\"Invalid simple block %r\" % block)\r\nValueError: Invalid simple block '39f236ea18'\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

ivankravets, 2020-11-04:
Fixed in the latest PIO Core. Please update

closed 2020-11-04 by ivankravets

ergerd, 2020-11-04:
I am on Core 5.0.3a2 is this not the latest version?