platformio/platformio-home #1735: Could not initialize project

maddentist, 2020-11-20:
PIO Core Call Error: "DLL load failed: The specified procedure could not be found."

maddentist, 2020-11-20:
--------------------------  ------------------------------------------------------
PlatformIO Core             5.0.3
Python                      3.7.7-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\Mathi\.platformio
PlatformIO Core Executable  C:\Users\Mathi\.platformio\penv\Scripts\platformio.exe
Python Executable           c:\users\mathi\.platformio\penv\scripts\python.exe
Global Libraries            0
Development Platforms       0
Tools & Toolchains          2
--------------------------  ------------------------------------------------------

ENG-HadiSamour, 2020-11-21:
I Have exactly the same probelm

ivankravets, 2020-12-02:
Duplicate of https://github.com/platformio/platformio-home/issues/1772

closed 2020-12-02 by ivankravets