platformio/platformio-home #1746: Could not load platform updates

AhmedAfifi2018, 2020-11-22:
PIO Core Call Error: "(2, 'CreateProcess', 'The system cannot find the file specified.')"

miftachuda, 2020-12-06:
same problem here, have you solve that problem?

YassineArrad, 2020-12-16:
same problem here

closed 2021-01-29 by stale[bot]

ColoMAX, 2021-02-06:
similar error with espressif8266, here with the Traceback:

antivirus disabled
and run as administrator 
PlatformIO Core             5.1.1a1
Python                      3.7.7-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252

>pio update          or similair with     >pio platform update espressif8266
Updating platformio/contrib-piohome           3.3.3 @ ~3.3.3                     [Up-to-date]
Updating platformio/contrib-pysite            2.37.191020 @ ~2.37.0              [Incompatible 2.39.201019]
Updating platformio/tool-unity                1.20500.200612 @ ~1.20500.0        [Up-to-date]
Updating platformio/tool-scons                4.40100.2 @ ~4.40100.2             [Up-to-date]
Updating platformio/tool-cppcheck             1.230.0 @ ~1.230.0                 [Up-to-date]

Platform Manager
================
Platform atmelavr
--------
Updating platformio/atmelavr                  3.1.0                              [Up-to-date]
Updating platformio/toolchain-atmelavr        1.50400.190710 @ ~1.50400.0        [Up-to-date]
Updating platformio/framework-arduino-avr     5.1.0 @ ~5.1.0                     [Up-to-date]
Updating platformio/framework-arduino-avr-attiny 1.3.2 @ ~1.3.2                     [Up-to-date]
Updating platformio/framework-simba           15.0.3 @ >=7.0.0                   [Up-to-date]
Updating platformio/tool-avrdude              1.60300.200527 @ ~1.60300.0        [Up-to-date]
Updating platformio/tool-simavr               1.10600.200511 @ ~1.10600.200511   [Up-to-date]

Platform espressif8266
--------
Updating git+https://github.com/platformio/platform-espressif8266.git 2.3.1                              Error: Traceback (most recent call last):
  File "c:\users\c010n\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\users\c010n\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\c010n\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\c010n\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\c010n\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\c010n\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\c010n\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\c010n\.platformio\penv\lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\c010n\.platformio\penv\lib\site-packages\platformio\commands\update.py", line 53, in cli
    ctx.invoke(cmd_platform_update, only_check=only_check)
  File "c:\users\c010n\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\c010n\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 421, in platform_update
    platform, only_packages=only_packages, only_check=only_check, silent=silent
  File "c:\users\c010n\.platformio\penv\lib\site-packages\platformio\package\manager\platform.py", line 113, in update
    show_incompatible=show_incompatible,
  File "c:\users\c010n\.platformio\penv\lib\site-packages\platformio\package\manager\_update.py", line 108, in update
    outdated = self.outdated(pkg, to_spec)
  File "c:\users\c010n\.platformio\penv\lib\site-packages\platformio\package\manager\_update.py", line 46, in outdated
    latest = self._fetch_vcs_latest_version(pkg)
  File "c:\users\c010n\.platformio\penv\lib\site-packages\platformio\package\manager\_update.py", line 69, in _fetch_vcs_latest_version
    vcs = VCSClientFactory.new(pkg.path, pkg.metadata.spec.url, silent=True)
  File "c:\users\c010n\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 55, in new
    src_dir, remote_url, tag, silent
  File "c:\users\c010n\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 137, in __init__
    self.configure()
  File "c:\users\c010n\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 146, in configure
    result = proc.exec_command([cls.command, "--exec-path"])
  File "c:\users\c010n\.platformio\penv\lib\site-packages\platformio\proc.py", line 117, in exec_command
    p = subprocess.Popen(*args, **kwargs)
  File "C:\Users\C010N\.platformio\python3\lib\subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "C:\Users\C010N\.platformio\python3\lib\subprocess.py", line 1207, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] The system cannot find the file specified