platformio/platformio-home #1768: Could not initialize project

Hendrikus-Simplex, 2020-11-27:
PIO Core Call Error: "The current working directory C:\\Users\\Hendrik Horn\\Documents\\PlatformIO\\Projects\\rx will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\Hendrik Horn\\Documents\\PlatformIO\\Projects\\rx\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing espmxdevkit (platform: espressif8266; board: espmxdevkit; framework: arduino)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nError: Unknown board ID 'espmxdevkit'\r\n========================== [FAILED] Took 0.68 seconds =========================="

kam9426, 2020-11-28:
I got the same error but my board is nucleo F103 RB

ivankravets, 2021-01-26:
Duplicate of https://github.com/platformio/platformio-home/issues/1768

closed 2021-01-26 by ivankravets

get2yomi, 2021-03-27:
PIO Core Call Error: "The current working directory C:\\Users\\galooli\\Documents\\PlatformIO\\Projects\\test will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\galooli\\Documents\\PlatformIO\\Projects\\test\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Unknown board ID 'genericSTM32F411CE'"

ivankravets, 2021-04-01:
@get2yomi please update all dev-platforms or run `pio update` command. Restart VSCode.

Bighoneypot, 2022-01-19:
@ivankravets 

You're up-to-date!
PlatformIO 5.2.4 is currently the newest version available.

PIO Core Call Error: "The next files/directories have been created in C:\\Users\\gabri\\Documents\\Platformio\\Projects\\Agrumino_Lemon_Captive_Portal\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing agruminolemon (platform: espressif8266; board: agruminolemon; framework: arduino)\r\n--------------------------------------------------------------------------------\r\nTool Manager: Installing framework-arduinoespressif8266 @ ~3.20704.0\r\nTool Manager: Warning! More than one package has been found by framework-arduinoespressif8266 @ ~3.20704.0 requirements:\r\n - platformio/framework-arduinoespressif8266 @ 3.30002.0\r\n - tasmota/framework-arduinoespressif8266 @ 2.7.4.9\r\nTool Manager: Please specify detailed REQUIREMENTS using package owner and version (shown above) to avoid name conflicts\r\nDownloading\r\nUnpacking\r\nTool Manager: framework-arduinoespressif8266 @ 3.20704.0 has been installed!\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nError: Unknown board ID 'agruminolemon'\r\n========================= [FAILED] Took 15.53 seconds ========================="

guozhaowen, 2022-06-18:
PIO Core Call Error: "Resolving upesy_wroom environment packages...\r\n\n\nError: Unknown board ID 'upesy_wroom'"

hneel, 2022-06-29:
Error: Unknown board ID 'lolin_s2_mini'

gotfredsen, 2022-07-06:
Core 6.0.2 - Home 3.4.2:
```
Could not initialize project
PIO Core Call Error: 
"Resolving **adafruit_qtpy_esp32c3** environment packages...
Error: Unknown board ID 'adafruit_qtpy_esp32c3'"
```

gotfredsen, 2022-07-06:
> Core 6.0.2 - Home 3.4.2:
> 
> ```
> Could not initialize project
> PIO Core Call Error: 
> "Resolving adafruit_qtpy_esp32c3 environment packages...
> Error: Unknown board ID 'adafruit_qtpy_esp32c3'"
> ```

Solved: 
- Uninstall Platform -> Espressif 32
- Install Platform -> Embedded -> Espressif 32
**No Automatic Update - Or reminder/reference to update needed**

ivankravets, 2022-07-06:
You use outdated dev-platform. Open PlatformIO IDE for VSCode > PlatformIO > Project Tasks > Update Dependencies

voidlaobai, 2022-07-11:
esp32-S3 can not be found
solution ——> VScode > PlantformIO home > plantforms > update

mahboud, 2022-08-05:
No matter what board I select I get:
> Error: Unknown board ID 'adafruit_qt_py_esp32-c3
or
> Error: Unknown board ID 'esp32-c3-mini

I've already done:

pio pkg update
```Resolving esp32dev dependencies...
Library Manager: Updating Adafruit MPU6050 @ 2.1.0+sha.1203a59
git version 2.21.0 (Apple Git-122.2)
Already up to date.
Library Manager: Updating WiFiManager @ 2.0.10-beta+sha.7d498ed
git version 2.21.0 (Apple Git-122.2)
Already up to date.
Already up-to-date.
Resolving esp32_c3_devkitc_02 dependencies...
Library Manager: Updating Adafruit MPU6050 @ 2.1.0+sha.1203a59
git version 2.21.0 (Apple Git-122.2)
Already up to date.
Library Manager: Updating WiFiManager @ 2.0.10-beta+sha.7d498ed
git version 2.21.0 (Apple Git-122.2)
Already up to date.
Already up-to-date.
Resolving esp32_c3_devkitm_1 dependencies...
Library Manager: Updating Adafruit MPU6050 @ 2.1.0+sha.1203a59
git version 2.21.0 (Apple Git-122.2)
Already up to date.
Library Manager: Updating WiFiManager @ 2.0.10-beta+sha.7d498ed
git version 2.21.0 (Apple Git-122.2)
Already up to date.
Already up-to-date.
Resolving adafruit_feather_esp32_v2 dependencies...
Library Manager: Updating Adafruit MPU6050 @ 2.1.0+sha.1203a59
git version 2.21.0 (Apple Git-122.2)
Already up to date.
Library Manager: Updating WiFiManager @ 2.0.10-beta+sha.7d498ed
git version 2.21.0 (Apple Git-122.2)
Already up to date.
Already up-to-date.```

Any thoughts?

ivankravets, 2022-08-05:
@mahboud please report to https://community.platformio.org/

Javesun99, 2022-11-16:
> 
Need to update Platform dependences not only libraries

frohro, 2022-11-22:
I can't find anything that needs updated, but I get this when I try to use VScode and PlatformIO to import working Arduino project or make a new project using the Wemos D1 R32 board which uses the Expressif ESP32 framework.  Everything says it is updated.

Rincewind08, 2022-11-30:
Having this very problem using the BTT distribution for the Octopus V1.1 tonight, very frustrated,  the main marlin branch doesn't work after it compiles, the screen freezes upon plugging in the EXP 2 connector and doesn't show up at all to the touchscreen interface of my TFT70

diamondbarcode, 2023-02-10:
it have been 3 years from first open of this topic I am wonder is there solution yet ? i just encounter this error today ..

exipilis, 2023-03-03:
The same error for me:
....
....
....
Error: Unknown board ID 'rpipico'"

frohro, 2023-03-27:
@gotfredsen had the solution for me.  
Thanks!
Rob

steve-barth, 2023-04-13:
Hi Ivan!
Please help, the next error message:
" Could not initialize project
PIO Core Call Error: "Resolving lilygo-t-display-s3 dependencies...\r\n\n\nError: Unknown board ID 'lilygo-t-display-s3'" "
( pio update, restart done )
Thanks, Steve

mariusl, 2023-04-16:
This error only came up after the system updated a number of extensions. Never had this problem before until yesterday.

`Could not initialize project
PIO Core Call Error: "Resolving lolin_s2_mini dependencies...\r\n\n\nError: Unknown board ID 'lolin_s2_mini'"`

mariusl, 2023-04-16:
Got the same error but only after updating all the extensions to the latest .

`Could not initialize project
PIO Core Call Error: "Resolving lolin_s2_mini dependencies...\r\n\n\nError: Unknown board ID 'lolin_s2_mini'"`

AgentKingNO47, 2023-05-18:
Got the same erro,how to fix this

GitLang, 2023-06-11:
OK, this is driving me mad!. I uninstalled all VSCode extensions and restarted it. Looked ok, so closed it and uninstalled vscode. Deleted all knows directories of vscode and extensions. Rebooted. Installed fresh latest VSCode from Microsoft. Installed PlatformIO extension and presumably it installed the latest it could of itself and dependencies. Installed latest PIO extension and restarted vscode. Clicked on PIO icon in right task bar. Clicked on Open :  Quick Access -> PIO Home -> Open. Got PIO window, congratulating me on having the latest 2023 version. Clicked on New Project in PIO window. Filled in project name, selected board and location and hit finish. AFter 5 seconds or so an error dialog appeared :

**PIO Core Call Error: "The following files/directories have been created in G:\\projects\\simple-foc-test-1\r\ninclude - Put project header files here\r\nlib - Put project specific (private) libraries here\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\nResolving disco_b_g431b_esc1 dependencies...\r\n\n\nError: Unknown board ID 'disco_b_g431b_esc1'**

I clicked on the suggested solutions button in the error dialog, it opened my usual browser with this page on view. It seems that whatever board I try to use from the list of 1500+ boards, I get a similar error and cannot proceed with even making a project. This thread and it's many duplicated shows this is an ongoing problem, and one that is stopping people proceeding with using PIO. Please, is there or can there be a solution?. Thank you.

jeffproe, 2023-06-21:
I just dusted off PlatformIO myself.  I hadn't touched it in a year or so and was getting basically the same error you were getting @GitLang.  I fixed it by going to `Platforms` and then `Updates` and running any updates that showed there.  Hope this helps!
[image]

omicronns, 2024-10-19:
Same problem, fresh windows + vscode + pio plugin install.
[image]
Same project works without problem on linux machine.

DMatheja, 2026-01-05:
Yes you need to update, its 2026 and still this error came up on my machine, maybe a fresh install would have also solved it, but the Update worked, idk why I had to search for this though!