platformio/platformio-home #1772: PIO Core Call Error: "DLL load failed: The specified procedure could not be found."

thijsdelft, 2020-11-28:
Could not load platform updates
PIO Core Call Error: "DLL load failed: The specified procedure could not be found."

thijsdelft, 2020-11-28:
The same happens when:
- adding a library to a project
- initializing a new project

Both in Atom and Visual Code Studio

From the command line interface platform updates *can* be installed.

Any hints on how to resolve this issue?

ivankravets, 2020-12-02:
Do you have multiple Pythons in a system?

thijsdelft, 2020-12-02:
Yes I do. (But I have this configuration already for quite some time)

thijsdelft, 2020-12-02:
After removing Anaconda the issue with the DLL procedure was resolved. 
I can now create projects and load platform updates again!

closed 2020-12-02 by thijsdelft

ivankravets, 2020-12-02:
See https://stackoverflow.com/questions/43264773/pil-dll-load-failed-specified-procedure-could-not-be-found

Yes, it is caused by anaconda. I don't people who use anaconda. They provide unofficial binaries of Python with unofficial libraries. What is the benefit? The whole PyPi registry is not compatible with their binaries.

Please use official Python and you will save your time and nerves 😊

maddentist, 2020-12-03:
No "broken" or unofficial Python dsitribution is installed on my system, and still I get the error. Please advise

ivankravets, 2020-12-03:
@maddentist Do you have any Python installed in your system?

TimSet0815, 2021-01-11:
I just uninstalled all versions of python and anaconda (which I did not had installed) and only freshly installed python 3.9.. I still get the DLL-error when checking for updates of my platforms. Any ideas?

ivankravets, 2021-01-20:
Resolved with porting PIO Home backend server to native Python 3 Async I/O. PlatformIO Core 5.0.5 will be released soon.