platformio/platformio-home #1785: Could not load platform data

AntonMalygin, 2020-12-01:
PIO Core Call Error: "Error: Could not find one of 'package.json' manifest files in the package"

ivankravets, 2021-01-26:
Please remove a broken package that does not contain a manifest file.

closed 2021-01-26 by ivankravets

HamzaHajeir, 2021-06-21:
How can we build with a specific version of a framework, such as:
```ini
platform = https://github.com/esp8266/Arduino/archive/78e63280ba2f736c1803bab6e82ad5a30492fc70.zip
```

Where that doesn't contain `platform.json` file?

ivankravets, 2021-06-22:
@HamzaHajeir you need to use https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform-packages

AntonMalygin, 2021-06-22:
Thanks for help for me..

> 
> 
>@HamzaHajeir you need to use  https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform-packages

neilmendoza, 2022-05-28:
I think I ended up with a partially downloaded microcontroller platform somehow and had to delete it from the .platformio folder to fix.

neilmendoza, 2022-05-30:
@DevlshOne I uninstalled Platformio, deleted the whole **YOUR_HOME_FOLDER/.platformio** folder and reinstalled in the end. Probably didn't need to go that far and just deleting the **YOUR_HOME_FOLDER/.platformio/platforms/MICROCONTOLLER_YOUR_HAVING_ISSUES_WITH** folder will probably work. **YOUR_HOME_FOLDER/.platformio** will probably be hidden, so on Mac you have to select go to folder from the Finder menu and Windows you have to show hidden folder.

rhousedorf, 2022-11-11:
this worked, just exiting vscode, deleting .platformio, running vscode again, and letting it install platformio again by itself.

ajsb85, 2022-12-11:
Very frustrating.
Nor reinstalling from scratch the OS.
I am using:

```
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.5 LTS
Release:        20.04
Codename:       focal
```

```
PLATFORM: Espressif 32 (5.2.0) > SparkFun ESP32 MicroMod
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40201.0 (4.2.1) 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 1.22851.191205 (2.28.51) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
```

curie31, 2023-07-14:
> this worked, just exiting vscode, deleting .platformio, running vscode again, and letting it install platformio again by itself.

Worked for me as well. Thanks!

walwer, 2023-12-05:
## TLDR
* Remove corrupted packages from `esphome` docker container and let them be downloaded again

---

I stepped upon the same issue recently inside Home Assistant environment and I just figured out the solution for that one.

## Description of issue
No possibility to install/build/compile from any configuration that uses `arduino` framework.
Example of failing configuration:
```yaml
esp32:
  board: esp-wrover-kit
  framework:
    type: arduino
```

## Cause of the issue in my case
Loss of network on HA environment caused incomplete download of package during the build.

## Solution
Steps to take:
1. Make sure you have terminal access to your HA instance. (I used https://github.com/hassio-addons/addon-ssh - Advanced SSH & Web Terminal)
2. If using above mentioned addon - make sure to disable "Safe Mode" under addon's setting. (HA Settings -> Addons -> Advanced SSH & Web Terminal -> Secure/Safe Mode -> Disable
3. Open terminal
4. List docker containers running on HA instance `docker ps`
5. The output should look more less like this:
[image]

6. Copy name of `esphome` container
7. Enter shell of `esphome` container
``` bash
docker exec -it <container_name or container id> /bin/bash
```
8. Navigate to following path
```
cd /data/cache/platformio/packages
```
9. If you list the files under that directory you should see some packages
```
***
framework-arduinoespressif32/
framework-arduinoespressif8266/
***
```
10. Remove any arduino related directories there
```bash
rm -r framework_arduino*
```
11. Try installing / building / compiling your configuration YAML file again.

The following solution worked for me and redownloaded the corrupted arduino build tool packages.
I hope this helps someone with similiar issue 😄

digimbyte, 2025-08-02:
still happening, package should auto detect and self resolve or have an option to purge - I cannot find any .platformio folder in my home directory so its some where else. 
fantastic.

All I did was try to create a new project since my other one was broken/corrupted.

[image]