platformio/platformio-home #1830: OS_MAKE_DIRS

tyeth, 2020-12-06:
Server error: {"type":"PermissionError","args":[13,"Permission denied"],"message":"[Errno 13] Permission denied: '/home/pi\\\\Documents\\\\test'"}

Cant open my project folder because its using incorrect slashes

tyeth, 2020-12-08:
My bad, think this is because I'm using code-insiders on windows10 but remote ssh plugin to access files. Then I tried remote-upload and it obviously (now) uses my system defaults as it tries to connect to a remote platformio system.

tyeth, 2020-12-08:
Would appreciate someone looking at this as it seems like a common scenario (remote development)

cdmistman, 2020-12-10:
As mentioned in platformio/platformio-vscode-ide#2209 there is a bug where PlatformIO Home uses Windows path separators instead of Linux separators when using WSL. This results in Projects being unrecognized in the pio Home page, and the inability to add these projects manually.

In the linked discussion, it was mentioned that [app/config.js](https://github.com/platformio/platformio-home/blob/fc0847e602fbf041e99afb384d4f0a16960e11e8/app/config.js#L17) might be getting the platform of the window that's running PlatformIO, which could be defined as `Win32` since the VSCode window is being run on Windows regardless of whether or not it's using remote development. This would explain #1094 and #1211 (as well as many other issues linked by users I won't link to prevent spamming), both of which seem to be running into problems caused by PlatformIO Home thinking it's running on Windows, when in fact it should be treating the remote environment as the execution environment.

In one of the mentioned issues, platformio/platformio-core#3681 was mentioned. I think it's also worth mentioning here.

Was gonna open a new issue but this one seems appropriate to avoid duplicates :)

lightmaster, 2021-01-24:
Just ran into this as well, no updating on fixing this bug?

closed 2021-01-31 by stale[bot]