platformio/platformio-home #1891: Could not load installed platforms

Spider84, 2020-12-16:
PIO Core Call Error: "Error: Unknown development platform 'PackageItem <path=C:\\Users\\spide\\.platformio\\platforms\\ststm32@9.0.0 metadata=PackageMetaData <type=platform name=ststm32 version=9.0.0 spec={'owner': 'platformio', 'id': 8020, 'name': 'ststm32', 'requirements': None, 'url': None}'"

ivankravets, 2021-01-20:
Duplicate of https://github.com/platformio/platformio-home/issues/2123

Could you provide more details on how to reproduce this issue?

closed 2021-01-20 by ivankravets

LucaPon, 2021-02-02:
Having the same problem, I tried to reinstall PlatformIO extension and to delete files in `%USERPROFILE%\.vscode\extensions` but it didn't help. The problem occurs when I click on "New project" in "PIO Home", when I open "Platforms" or "Boards" sections or when I click on "Updates"->"Platform Updates".

`Error: Unknown development platform 'PackageItem <path=C:\Users\Luca\.platformio\platforms\ststm32 metadata=PackageMetaData <type=platform name=ststm32 version=9.0.0 spec={'owner': 'platformio', 'id': 8020, 'name': 'ststm32', 'requirements': None, 'url': None}'`