platformio/platformio-home #1924: Could not initialize project

mledimna, 2020-12-21:
PIO Core Call Error: "The current working directory /home/mathieu-ledimna/Programmation/STM32/VSCode/Test-Build will be used for the project.\n\nThe next files/directories have been created in /home/mathieu-ledimna/Programmation/STM32/VSCode/Test-Build\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing disco_f746ng (platform: ststm32; board: disco_f746ng; framework: mbed)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm32/disco_f746ng.html\nPLATFORM: ST STM32 (10.0.1) > ST 32F746GDISCOVERY\nHARDWARE: STM32F746NGH6 216MHz, 320KB RAM, 1MB Flash\nDEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)\nPACKAGES: \n - framework-mbed 6.60200.200722 (6.2.0) \n - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)\nAttributeError: type object 'array.array' has no attribute 'tostring':\n  File \"/home/mathieu-ledimna/.platformio/penv/lib/python3.9/site-packages/platformio/builder/main.py\", line 170:\n    env.SConscript(\"$BUILD_SCRIPT\")\n  File \"/home/mathieu-ledimna/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py\", line 598:\n    return _SConscript(self.fs, *files, **subst_kw)\n  File \"/home/mathieu-ledimna/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py\", line 287:\n    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\n  File \"/home/mathieu-ledimna/.platformio/platforms/ststm32/builder/main.py\", line 99:\n    target_elf = env.BuildProgram()\n  File \"/home/mathieu-ledimna/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py\", line 219:\n    return self.method(*nargs, **kwargs)\n  File \"/home/mathieu-ledimna/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/platformio.py\", line 61:\n    env.ProcessProgramDeps()\n  File \"/home/mathieu-ledimna/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py\", line 219:\n    return self.method(*nargs, **kwargs)\n  File \"/home/mathieu-ledimna/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/platformio.py\", line 127:\n    env.BuildFrameworks(env.get(\"PIOFRAMEWORK\"))\n  File \"/home/mathieu-ledimna/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py\", line 219:\n    return self.method(*nargs, **kwargs)\n  File \"/home/mathieu-ledimna/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/platformio.py\", line 342:\n    SConscript(env.GetFrameworkScript(f), exports=\"env\")\n  File \"/home/mathieu-ledimna/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py\", line 661:\n    return method(*args, **kw)\n  File \"/home/mathieu-ledimna/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py\", line 598:\n    return _SConscript(self.fs, *files, **subst_kw)\n  File \"/home/mathieu-ledimna/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py\", line 287:\n    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\n  File \"/home/mathieu-ledimna/.platformio/platforms/ststm32/builder/frameworks/mbed.py\", line 35:\n    SConscript(\n  File \"/home/mathieu-ledimna/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py\", line 661:\n    return method(*args, **kw)\n  File \"/home/mathieu-ledimna/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py\", line 598:\n    return _SConscript(self.fs, *files, **subst_kw)\n  File \"/home/mathieu-ledimna/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py\", line 287:\n    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\n  File \"/home/mathieu-ledimna/.platformio/packages/framework-mbed/platformio/platformio-build.py\", line 43:\n    from pio_mbed_adapter import PlatformioMbedAdapter\n  File \"/home/mathieu-ledimna/.platformio/packages/framework-mbed/platformio/pio_mbed_adapter.py\", line 23:\n    from tools.build_api import prepare_toolchain, UPDATE_WHITELIST\n  File \"/home/mathieu-ledimna/.platformio/packages/framework-mbed/tools/build_api.py\", line 37:\n    from .utils import (mkdir, run_cmd, run_cmd_ext, NotSupportedException,\n  File \"/home/mathieu-ledimna/.platformio/packages/framework-mbed/tools/utils.py\", line 33:\n    from intelhex import IntelHex\n  File \"/home/mathieu-ledimna/.platformio/packages/framework-mbed/platformio/package_deps/py3/intelhex/__init__.py\", line 44:\n    from intelhex.compat import (\n  File \"/home/mathieu-ledimna/.platformio/packages/framework-mbed/platformio/package_deps/py3/intelhex/compat.py\", line 60:\n    array_tobytes = getattr(array.array, \"tobytes\", array.array.tostring)\n========================== [FAILED] Took 0.41 seconds =========================="

mledimna, 2020-12-21:
The problem was from the mbed framework :

.platformio/packages/framework-mbed/platformio/package_deps/py3/intelhex/compat.py", line 60

Change array_tobytes = getattr(array.array, “tobytes”, **array.array.tostring**)
To array_tobytes = getattr(array.array, “tobytes”, **array.array.tobytes**)

The tostring attribute is deprecated since python 3.9

ivankravets, 2021-01-26:
Fixed in https://github.com/python-intelhex/intelhex/commit/4125dce5b174401d38cc0fcf9d2e1aad07997f5e

closed 2021-01-26 by ivankravets