platformio/platformio-home #1930: Could not install platform

NeliusNDL, 2020-12-22:
PIO Core Call Error: "Platform Manager: Installing git+https://github.com/espressif/arduino-esp32.git\r\ngit version 2.28.0.windows.1\nSubmodule path 'libraries/AzureIoT': checked out '5e8ffb21115675f8c258e81bb287a9cd610e3205'\n\n\nCloning into 'C:\\Users\\Nelius\\.platformio\\.cache\\tmp\\pkg-installing-muzrt0cg'...\nUpdating files:  22% (468/2083)\rUpdating files:  22% (470/2083)\rUpdating files:  22% (472/2083)\rUpdating files:  23% (480/2083)\rUpdating files:  24% (500/2083)\rUpdating files:  25% (521/2083)\rUpdating files:  26% (542/2083)\rUpdating files:  27% (563/2083)\rUpdating files:  28% (584/2083)\rUpdating files:  29% (605/2083)\rUpdating files:  30% (625/2083)\rUpdating files:  31% (646/2083)\rUpdating files:  32% (667/2083)\rUpdating files:  33% (688/2083)\rUpdating files:  34% (709/2083)\rUpdating files:  35% (730/2083)\rUpdating files:  36% (750/2083)\rUpdating files:  37% (771/2083)\rUpdating files:  38% (792/2083)\rUpdating files:  39% (813/2083)\rUpdating files:  40% (834/2083)\rUpdating files:  41% (855/2083)\rUpdating files:  41% (872/2083)\rUpdating files:  42% (875/2083)\rUpdating files:  43% (896/2083)\rUpdating files:  44% (917/2083)\rUpdating files:  45% (938/2083)\rUpdating files:  46% (959/2083)\rUpdating files:  47% (980/2083)\rUpdating files:  48% (1000/2083)\rUpdating files:  49% (1021/2083)\rUpdating files:  50% (1042/2083)\rUpdating files:  51% (1063/2083)\rUpdating files:  52% (1084/2083)\rUpdating files:  53% (1104/2083)\rUpdating files:  54% (1125/2083)\rUpdating files:  55% (1146/2083)\rUpdating files:  56% (1167/2083)\rUpdating files:  57% (1188/2083)\rUpdating files:  58% (1209/2083)\rUpdating files:  59% (1229/2083)\rUpdating files:  60% (1250/2083)\rUpdating files:  61% (1271/2083)\rUpdating files:  62% (1292/2083)\rUpdating files:  63% (1313/2083)\rUpdating files:  63% (1314/2083)\rUpdating files:  64% (1334/2083)\rUpdating files:  65% (1354/2083)\rUpdating files:  66% (1375/2083)\rUpdating files:  67% (1396/2083)\rUpdating files:  68% (1417/2083)\rUpdating files:  69% (1438/2083)\rUpdating files:  70% (1459/2083)\rUpdating files:  71% (1479/2083)\rUpdating files:  72% (1500/2083)\rUpdating files:  73% (1521/2083)\rUpdating files:  74% (1542/2083)\rUpdating files:  75% (1563/2083)\rUpdating files:  76% (1584/2083)\rUpdating files:  77% (1604/2083)\rUpdating files:  78% (1625/2083)\rUpdating files:  79% (1646/2083)\rUpdating files:  80% (1667/2083)\rUpdating files:  80% (1675/2083)\rUpdating files:  81% (1688/2083)\rUpdating files:  82% (1709/2083)\rUpdating files:  83% (1729/2083)\rUpdating files:  84% (1750/2083)\rUpdating files:  85% (1771/2083)\rUpdating files:  86% (1792/2083)\rUpdating files:  87% (1813/2083)\rUpdating files:  88% (1834/2083)\rUpdating files:  89% (1854/2083)\rUpdating files:  90% (1875/2083)\rUpdating files:  91% (1896/2083)\rUpdating files:  92% (1917/2083)\rUpdating files:  92% (1922/2083)\rUpdating files:  93% (1938/2083)\rUpdating files:  94% (1959/2083)\rUpdating files:  95% (1979/2083)\rUpdating files:  96% (2000/2083)\rUpdating files:  97% (2021/2083)\rUpdating files:  97% (2040/2083)\rUpdating files:  98% (2042/2083)\rUpdating files:  99% (2063/2083)\rUpdating files: 100% (2083/2083)\rUpdating files: 100% (2083/2083), done.\nSubmodule 'libraries/AzureIoT' (https://github.com/VSChina/ESP32_AzureIoT_Arduino) registered for path 'libraries/AzureIoT'\nCloning into 'C:/Users/Nelius/.platformio/.cache/tmp/pkg-installing-muzrt0cg/libraries/AzureIoT'...\nError: Could not find one of 'platform.json' manifest files in the package"

NeliusNDL, 2020-12-22:
22 12 2020

ivankravets, 2021-01-26:
Duplicate of https://github.com/platformio/platformio-home/issues/1785

closed 2021-01-26 by ivankravets