platformio/platformio-home #1948: Libraries: Builtin

mklengel, 2020-12-24:
PIO Core Call Error: "b''\n\nError: Traceback (most recent call last):\n  File \"/Users/mklengel/.platformio/penv/lib/python3.8/site-packages/platformio/__main__.py\", line 109, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"/Users/mklengel/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 829, in __call__\n    return self.main(*args, **kwargs)\n  File \"/Users/mklengel/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 782, in main\n    rv = self.invoke(ctx)\n  File \"/Users/mklengel/.platformio/penv/lib/python3.8/site-packages/platformio/commands/__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"/Users/mklengel/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/mklengel/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/mklengel/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 1066, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"/Users/mklengel/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 610, in invoke\n    return callback(*args, **kwargs)\n  File \"/Users/mklengel/.platformio/penv/lib/python3.8/site-packages/platformio/commands/lib/command.py\", line 419, in lib_builtin\n    items = get_builtin_libs(storage)\n  File \"/Users/mklengel/.platformio/penv/lib/python3.8/site-packages/platformio/commands/lib/helpers.py\", line 34, in get_builtin_libs\n    for storage in p.get_lib_storages():\n  File \"/Users/mklengel/.platformio/penv/lib/python3.8/site-packages/platformio/platform/base.py\", line 214, in get_lib_storages\n    pkg = self.get_package(opts[\"package\"])\n  File \"/Users/mklengel/.platformio/penv/lib/python3.8/site-packages/platformio/platform/_packages.py\", line 31, in get_package\n    return self.pm.get_package(self.get_package_spec(name))\n  File \"/Users/mklengel/.platformio/penv/lib/python3.8/site-packages/platformio/platform/_packages.py\", line 21, in get_package_spec\n    version = self.packages[name].get(\"version\", \"\")\nKeyError: 'framework-stm32cube'\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

mklengel, 2020-12-24:
Selecting Libraries->Builtin leads to the error message.

ivankravets, 2021-01-26:
Fixed in the latest ST STM32 dev-platform. Please open `PIO Home > PlatformIO > Updates` and update to the latest version.

closed 2021-01-26 by ivankravets