platformio/platformio-home #1964: Could not load installed platforms

ManuelZ, 2020-12-27:
PIO Core Call Error: "Error: Unknown development platform 'PackageItem <path=C:\\Users\\Manuel\\.platformio\\platforms\\ststm32 metadata=PackageMetaData <type=platform name=ststm32 version=10.0.1 spec={'owner': 'platformio', 'id': 8020, 'name': 'ststm32', 'requirements': None, 'url': None}'"

I deleted my whole `.platformio` folder, then opened up VSCode again, and I found the STM32 platform installed, I clicked "uninstall", was a success, restarted VSCode and I found this message in a popup.

ivankravets, 2021-01-26:
Fixed in the latest version. Please run `pio update` and restart VSCode.

closed 2021-01-26 by ivankravets

Bluscream, 2022-04-01:
I have the same issue:

```
> Executing task: C:\Users\blusc\.platformio\penv\Scripts\platformio.exe run <

Processing esp12e (platform: espressif8266; board: d1_mini; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: espressif8266 @ 3.2.0 is already installed
Error: Unknown development platform 'PackageItem <path=C:\Users\blusc\.platformio\platforms\espressif8266 metadata=PackageMetaData <type=platform name=espressif8266 version=3.2.0 spec={'owner': 'platformio', 'id': 8008, 'name': 'espressif8266', 'requirements': None, 'url': None}'
The terminal process "C:\Users\blusc\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
```

ivankravets, 2022-04-01:
Could you reproduce this issue after restart of VSCode?

ivankravets, 2022-04-01:
Try to remove `C:\Users\blusc\.platformio\platforms\espressif8266` folder.

Bluscream, 2022-04-01:
That worked, thank you

ivankravets, 2022-04-01:
It seems that some spy software on your machine does bad work. Do you use any 3rd part "antivirus" tools?

aschen-builder, 2022-09-26:
> It seems that some spy software on your machine does bad work. Do you use any 3rd part "antivirus" tools?

Even though this is old, could you elaborate on what you meant? What stands out to you as being significant of malicious software?

rtek1000, 2023-04-01:
This 'Unknown development platform' error popped up here but it worked after trying to install with pip

I tried to compile this example:
https://github.com/platformio/platform-ststm8/tree/master/examples/spl-blink

https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html#local-download-macos-linux-windows
```
# change directory to the folder where is located downloaded "get-platformio.py"
cd /path-to-dir/where/get-platformio.py/is-located

# run it
python get-platformio.py
```

`pio run`

> Command 'pio' not found, but can be installed with:
> sudo apt install platformio

`sudo apt install platformio`

`pio run`

> ********************************************************************************
> Obsolete PIO Core v4.3.4 is used (previous was 6.1.6)
> Please remove multiple PIO Cores from a system:
> https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
> ********************************************************************************
> Processing stm8sdisco (platform: ststm8; framework: spl; board: stm8sdisco)
> --------------------------------------------------------------------------------
> PlatformManager: Installing ststm8
> ststm8 @ 2.1.0 is already installed
> Error: Unknown development platform 'ststm8'

`pio update`

> ********************************************************************************
> Obsolete PIO Core v4.3.4 is used (previous was 6.1.6)
> Please remove multiple PIO Cores from a system:
> https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
> ********************************************************************************
> Error: Traceback (most recent call last):
>   File "/usr/lib/python3/dist-packages/platformio/__main__.py", line 109, in main
>     cli()  # pylint: disable=no-value-for-parameter
>   File "/usr/lib/python3/dist-packages/click/core.py", line 1128, in __call__
>     return self.main(*args, **kwargs)
>   File "/usr/lib/python3/dist-packages/click/core.py", line 1053, in main
>     rv = self.invoke(ctx)
>   File "/usr/lib/python3/dist-packages/platformio/commands/__init__.py", line 44, in invoke
>     return super(PlatformioCLI, self).invoke(ctx)
>   File "/usr/lib/python3/dist-packages/click/core.py", line 1659, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "/usr/lib/python3/dist-packages/click/core.py", line 1395, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>   File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke
>     return __callback(*args, **kwargs)
>   File "/usr/lib/python3/dist-packages/click/decorators.py", line 26, in new_func
>     return f(get_current_context(), *args, **kwargs)
>   File "/usr/lib/python3/dist-packages/platformio/commands/update.py", line 45, in cli
>     update_core_packages(only_check)
>   File "/usr/lib/python3/dist-packages/platformio/managers/core.py", line 95, in update_core_packages
>     pkg_dir = pm.get_package_dir(name)
>   File "/usr/lib/python3/dist-packages/platformio/managers/package.py", line 415, in get_package_dir
>     manifest = self.get_package(name, requirements, url)
>   File "/usr/lib/python3/dist-packages/platformio/managers/package.py", line 384, in get_package
>     for manifest in self.get_installed():
>   File "/usr/lib/python3/dist-packages/platformio/managers/package.py", line 374, in get_installed
>     manifest = self.load_manifest(pkg_dir)
>   File "/usr/lib/python3/dist-packages/platformio/managers/package.py", line 347, in load_manifest
>     manifest = ManifestParserFactory.new_from_file(manifest_path).as_dict()
[... 229 lines cut ...]
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/stm8sblue/SPL/stm8s_gpio.rel
Compiling .pio/build/stm8sblue/src/main.rel
Compiling .pio/build/stm8sblue/src/stm8s_it.rel
src/main.c:112: warning 85: in function assert_failed unreferenced function argument : 'file'
src/main.c:112: warning 85: in function assert_failed unreferenced function argument : 'line'
Linking .pio/build/stm8sblue/firmware.elf
Checking size .pio/build/stm8sblue/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.0% (used 0 bytes from 1024 bytes)
Flash: [=         ]   8.1% (used 667 bytes from 8192 bytes)
Building .pio/build/stm8sblue/firmware.hex
========================= [SUCCESS] Took 0.75 seconds =========================

Environment     Status    Duration
--------------  --------  ------------
stm8sdisco      SUCCESS   00:00:00.803
s8uno           SUCCESS   00:00:00.742
mb208           SUCCESS   00:00:00.769
nucleo_8s207k8  SUCCESS   00:00:00.784
nucleo_8s208rb  SUCCESS   00:00:00.775
stm8sblue       SUCCESS   00:00:00.751
========================= 6 succeeded in 00:00:04.625 =========================

ivankravets, 2023-04-01:
> Obsolete PIO Core v4.3.4 is used

Please remove PlatformIO Core 4. See https://docs.platformio.org/en/latest/core/installation/troubleshooting.html#multiple-platformio-cores-in-a-system

rtek1000, 2023-04-01:
> > Obsolete PIO Core v4.3.4 is used
> 
> Please remove PlatformIO Core 4. See https://docs.platformio.org/en/latest/core/installation/troubleshooting.html#multiple-platformio-cores-in-a-system

> If you have [PlatformIO IDE](https://docs.platformio.org/en/latest/integration/ide/pioide.html#pioide) already installed, you do not need to install [PlatformIO Core (CLI)](https://docs.platformio.org/en/latest/core/index.html#piocore) separately

https://docs.platformio.org/en/latest/core/installation/shell-commands.html