platformio/platformio-home #2096: Could not initialize project

sandro-sole, 2021-01-17:
PIO Core Call Error: "The current working directory /Users/sandro/Documents/PlatformIO/Projects/esp32/esp_surv will be used for the project.\n\nThe next files/directories have been created in /Users/sandro/Documents/PlatformIO/Projects/esp32/esp_surv\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing esp32cam (platform: espressif32; board: esp32cam; framework: espidf)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html\nPLATFORM: Espressif 32 (2.1.0) > AI Thinker ESP32-CAM\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\nPACKAGES: \n - framework-espidf 3.40100.200827 (4.1.0) \n - tool-cmake 3.16.4 \n - tool-esptoolpy 1.30000.201119 (3.0.0) \n - tool-ninja 1.9.0 \n - toolchain-esp32ulp 1.22851.191205 (2.28.51) \n - toolchain-xtensa32 2.80200.200827 (8.2.0)\nReading CMake configuration...\nAssertionError: :\n  File \"/Users/sandro/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py\", line 170:\n    env.SConscript(\"$BUILD_SCRIPT\")\n  File \"/Users/sandro/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py\", line 598:\n    return _SConscript(self.fs, *files, **subst_kw)\n  File \"/Users/sandro/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py\", line 287:\n    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\n  File \"/Users/sandro/.platformio/platforms/espressif32/builder/main.py\", line 224:\n    target_elf = env.BuildProgram()\n  File \"/Users/sandro/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py\", line 219:\n    return self.method(*nargs, **kwargs)\n  File \"/Users/sandro/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/platformio.py\", line 61:\n    env.ProcessProgramDeps()\n  File \"/Users/sandro/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py\", line 219:\n    return self.method(*nargs, **kwargs)\n  File \"/Users/sandro/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/platformio.py\", line 127:\n    env.BuildFrameworks(env.get(\"PIOFRAMEWORK\"))\n  File \"/Users/sandro/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py\", line 219:\n    return self.method(*nargs, **kwargs)\n  File \"/Users/sandro/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/platformio.py\", line 342:\n    SConscript(env.GetFrameworkScript(f), exports=\"env\")\n  File \"/Users/sandro/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py\", line 661:\n    return method(*args, **kw)\n  File \"/Users/sandro/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py\", line 598:\n    return _SConscript(self.fs, *files, **subst_kw)\n  File \"/Users/sandro/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py\", line 287:\n    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\n  File \"/Users/sandro/.platformio/platforms/espressif32/builder/frameworks/espidf.py\", line 1177:\n    ota_partition_params = get_partition_info(\n  File \"/Users/sandro/.platformio/platforms/espressif32/builder/frameworks/espidf.py\", line 838:\n    assert os.path.isfile(pt_path)\n========================== [FAILED] Took 5.69 seconds =========================="

sandro-sole, 2021-01-17:
When I start a new project with the Arduino platform, it works. But as soon as I try the ESP-IDF platform, I'm facing this issue.
I also tried the solution described in https://github.com/platformio/platformio-core/issues/2517, but it did not fix the problem.
Any suggestions?

My System: osx, latest Plattformio

ivankravets, 2021-01-26:
Duplicate of https://github.com/platformio/platform-espressif32/issues/476

closed 2021-01-26 by ivankravets