platformio/platformio-home #2103: Could not initialize project

toliy1409, 2021-01-18:
PIO Core Call Error: "The current working directory C:\\Users\\Toliy\\Documents\\PlatformIO\\Projects\\project ESP_IDF will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\Toliy\\Documents\\PlatformIO\\Projects\\project ESP_IDF\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html\r\nPLATFORM: Espressif 32 (2.0.0) > Espressif ESP32 Dev Module\r\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\r\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\r\nPACKAGES: \r\n - framework-espidf 3.40100.200827 (4.1.0) \r\n - tool-cmake 3.16.4 \r\n - tool-esptoolpy 1.20600.0 (2.6.0) \r\n - tool-idf 1.0.1 \r\n - tool-mconf 1.4060000.20190628 (406.0.0) \r\n - tool-ninja 1.9.0 \r\n - toolchain-esp32ulp 1.22851.191205 (2.28.51) \r\n - toolchain-xtensa32 2.80200.200827 (8.2.0)\r\nError: Detected a whitespace character in project paths.\r\n========================== [FAILED] Took 2.35 seconds =========================="

toliy1409, 2021-01-18:
При запуске проекта на платформе Platformio с фрэймворк ESPIDF вылетает такая ошибка

ivankravets, 2021-01-20:
Duplicate of https://github.com/platformio/platform-espressif32/issues/470

closed 2021-01-20 by ivankravets