platformio/platformio-home #2123: Could not load boards list

1Speedi1, 2021-01-20:
PIO Core Call Error: "Error: Unknown development platform 'PackageItem <path=C:\\Users\\Kenneth Wermuth\\.platformio\\platforms\\atmelavr metadata=PackageMetaData <type=platform name=atmelavr version=3.1.0 spec={'owner': 'platformio', 'id': 8006, 'name': 'atmelavr', 'requirements': None, 'url': None}'"

ivankravets, 2021-01-20:
Could you provide more details on how to reproduce this issue?

LucaPon, 2021-02-03:
Having the same problem, I tried to reinstall PlatformIO extension and to delete files in `%USERPROFILE%\.vscode\extensions` but it didn't help. The problem occurs when I click on "New project" in "PIO Home", when I open "Platforms" or "Boards" sections or when I click on "Updates"->"Platform Updates".

`Error: Unknown development platform 'PackageItem <path=C:\Users\Luca\.platformio\platforms\ststm32 metadata=PackageMetaData <type=platform name=ststm32 version=9.0.0 spec={'owner': 'platformio', 'id': 8020, 'name': 'ststm32', 'requirements': None, 'url': None}'`

ivankravets, 2021-02-03:
@LucaPon Could you reproduce this issue when restarting VSCode?

LucaPon, 2021-02-03:
Yes, it happens every time. I tried restarting vscode, rebooting and reinstalling extension.

ivankravets, 2021-02-03:
Please attach here the contents of 2 files:

- `C:\Users\Luca\.platformio\platforms\ststm32\.piopm`
- `C:\Users\Luca\.platformio\platforms\ststm32\package.json`

It seems like dangerous 3rd party antivirus makes problems for you. Do you use them?

jatinkohli, 2021-02-03:
I was having the same issue and it was resolved for me by deleting the platform before building my project. For @LucaPon, that would be deleting the ststm32 folder in the 'C:\Users\Luca\.platformio\platforms\' folder. I had this issue when I changed computers and installed the VSCode platformio extension with having an existing platformio project currently in my workspace. I was also running VScode in administrator mode when I downloaded the extension, but I don't see how that would cause this.

ivankravets, 2021-02-03:
@jatinkohli thanks for your solution!

@LucaPon please provide the information requested above that could help us to fix this issue. We can't reproduce it on our side.

LucaPon, 2021-02-03:
@ivankravets at this link you can find the .piopm file: https://we.tl/t-jHT7LfC3G8 , I have no package.json in that folder. 
A thing that I noticed is that In platforms folder I have both ststm32 and ststm32@8.0.0 directories, but I don't know if it can be related to the problem. 
Anyway deleting ststm32 folder and restarting vscode solved the problem. Thank you @jatinkohli!!

Hades32, 2021-04-02:
I just run into the same issue and deleting the folder under `~/.platformio/platforms` sure helped. I did a quick file listing comparison and the "bad" version was missing dozends of files. So it seems a previous install attempt failed.

From previous experience I would suggest to introduce a marker file to indicate that a platform has been completely extracted. (Renaming/moving the folder might sound like the better solution, but that quickly get's you into trouble under Windows with locked files due to anti-virus tools...)

danburkhol, 2022-06-16:
☝️ Same as above, I was updating my platforms and libraries, after not doing so in a while, and the moment I clicked the update button to update my Teensy platform, I started getting this problem. Uninstalled and deleted `~/.platformio`, re-installed, and no more error.

Gonkers, 2022-06-18:
Same as most everyone above. Thanks @jatinkohli 
I deleted ~/.platformio and reinstalled the extension to get it working again.

navdotnetreqs, 2022-06-20:
Not too pleased about a bug that can only be fixed by deleting folders and reinstalling platformio, but that did fix the issue for me too.

IoakeimSogiakas, 2022-06-29:
> I was having the same issue and it was resolved for me by deleting the platform before building my project. For @LucaPon, that would be deleting the ststm32 folder in the 'C:\Users\Luca.platformio\platforms' folder. I had this issue when I changed computers and installed the VSCode platformio extension with having an existing platformio project currently in my workspace. I was also running VScode in administrator mode when I downloaded the extension, but I don't see how that would cause this.
> 
> Edit: For me, just a simple restart of VSCode did not fix the issue.

bless u

nomis, 2022-07-02:
In my case the problem is caused by `espressif32@5.0.0` being incompatible with platformio 5.2.5. This would happen to anyone specifying the version as `espressif32` upon updating to the latest version of the platform but not the latest version of platformio.

```
$ platformio --version
PlatformIO Core, version 5.2.5
$ platformio run 
Processing s2_mini (platform: espressif32@^5.0.0; board: lolin_s2_mini; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: espressif32 @ 5.0.0 is already installed
Error: Unknown development platform 'espressif32'
```

The code in `platformio/platform/factory.py` reports "unknown platform" if there's an import error, which can be caused by the incompatible platformio API:

```
class PlatformFactory(object):
    @staticmethod
    def load_module(name, path):
        try:
            return load_python_module("platformio.platform.%s" % name, path)
        except ImportError:
            raise UnknownPlatform(name)
```

```
$ platformio --version
PlatformIO Core, version 5.2.5
$ platformio run 
Processing s2_mini (platform: espressif32@^5.0.0; board: lolin_s2_mini; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: espressif32 @ 5.0.0 is already installed
Error: Unknown development platform 'espressif32'
```

If I modify `PlatformFactory` to show the exception, it's failing in `espressif32@5.0.0/platform.py`:
```
$ platformio run
Processing s2_mini (platform: espressif32@^5.0.0; board: lolin_s2_mini; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Traceback (most recent call last):
  File "/home/simon/.local/lib/python3.6/site-packages/platformio/__main__.py", line 121, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/simon/.local/lib/python3.6/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/home/simon/.local/lib/python3.6/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/home/simon/.local/lib/python3.6/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/simon/.local/lib/python3.6/site-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/simon/.local/lib/python3.6/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/simon/.local/lib/python3.6/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/home/simon/.local/lib/python3.6/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/simon/.local/lib/python3.6/site-packages/platformio/commands/run/command.py", line 144, in cli
    is_test_running,
  File "/home/simon/.local/lib/python3.6/site-packages/platformio/commands/run/command.py", line 176, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "/home/simon/.local/lib/python3.6/site-packages/platformio/commands/run/processor.py", line 67, in process
    result = init_platform(self.options["platform"]).run(
  File "/home/simon/.local/lib/python3.6/site-packages/platformio/commands/platform.py", line 322, in init_platform
    return PlatformFactory.new(name)
  File "/home/simon/.local/lib/python3.6/site-packages/platformio/platform/factory.py", line 73, in new
    platform_name, os.path.join(platform_dir, "platform.py")
  File "/home/simon/.local/lib/python3.6/site-packages/platformio/platform/factory.py", line 34, in load_module
    return load_python_module("platformio.platform.%s" % name, path)
  File "/home/simon/.local/lib/python3.6/site-packages/platformio/compat.py", line 64, in load_python_module
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/simon/.platformio/platforms/espressif32/platform.py", line 22, in <module>
    from platformio.public import PlatformBase, to_unix_path
ModuleNotFoundError: No module named 'platformio.public'
```

TurnOffNOD, 2022-09-08:
Hi everyone, my problem occurred when platform itself was installed into default location (windows %USERPROFILE%\.platformio), but set platform and other tools installation dir to d:\<my dir location> with PLATFORMIO_PLATFORMS_DIR and PLATFORMIO_PACKAGES_DIR.

I uninstalled old installation, removed these environment variables, and set installation location to d:\<my dir location> with PLATFORMIO_CORE_DIR, the problem solved.

ngosangns, 2025-01-24:
In my case. I just want to delete the temp folders inside `%USERPROFILE%/.platformio\platforms`