platformio/platformio-home #2124: Could not load installed platforms

1Speedi1, 2021-01-20:
PIO Core Call Error: "Error: Unknown development platform 'PackageItem <path=C:\\Users\\Kenneth Wermuth\\.platformio\\platforms\\atmelavr metadata=PackageMetaData <type=platform name=atmelavr version=3.1.0 spec={'owner': 'platformio', 'id': 8006, 'name': 'atmelavr', 'requirements': None, 'url': None}'"

ivankravets, 2021-01-20:
Duplicate of https://github.com/platformio/platformio-home/issues/2123

Could you provide more details on how to reproduce this issue?

closed 2021-01-20 by ivankravets

1Speedi1, 2021-01-21:
> Duplicate of #2123
> 
> Could you provide more details on how to reproduce this issue?

Sorry, Iam new, so did not know I posted it in forum.  
It came up when I tried to select board, so could not select any board.