platformio/platformio-home #2144: Could not install platform

osc-github, 2021-01-22:
PIO Core Call Error: "Platform Manager: Installing espressiv32 @ 1.10.0\r\n\n\nError: Could not find the package with 'espressiv32 @ 1.10.0' requirements for your system 'windows_amd64'"

ivankravets, 2021-01-22:
What is `espressiv32`??? The correct list of development platforms could be found at https://platformio.org/platforms

closed 2021-01-22 by ivankravets

Hancke, 2023-06-06:
Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r1' requirements for your system 'windows_amd64'"

Strat00s, 2023-08-13:
Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3' requirements for your system 'windows_amd64'"

Doubletop12, 2023-09-25:
Project Configuration File
Resolving adafruit_feather_esp32s3 dependencies...
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3

UnknownPackageError: Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3' requirements for your system 'windows_amd64'"

Doubletop12, 2023-09-25:
Fixed - see here

https://github.com/platformio/platform-espressif32/issues/1081

gallochri, 2023-10-06:
PIO Core Call Error: "The following files/directories have been created in /home/gallochri/NextCasa/home/gallochri/Documenti/PlatformIO/Projects/test2\ninclude - Put project header files here\nlib - Put project specific (private) libraries here\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\nResolving esp32dev dependencies...\nTool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch2\n\n\nUnknownPackageError: Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch2' requirements for your system 'linux_x86_64'"

Alex-IC, 2024-02-17:
PIO Core Call Error: "Platform Manager: Installing logicrom\r\nUnpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%\r\nPlatform Manager: logicrom@0.9.7 has been installed!\r\nTool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.80201.0\r\nDownloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%\r\nUnpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%\r\nTool Manager: toolchain-gccarmnoneeabi@1.80201.181220 has been installed!\r\nTool Manager: Installing waybyte/framework-logicromsdk @ ~0.9.9\r\n\n\nUnknownPackageError: Could not find the package with 'waybyte/framework-logicromsdk @ ~0.9.9' requirements for your system 'windows_amd64'"

diszell2008, 2024-05-06:
PIO Core Call Error: "The following files/directories have been created in /Users/diszell2008/Documents/PlatformIO/Projects/ESP32s3\ninclude - Put project header files here\nlib - Put project specific (private) libraries here\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\nResolving esp32-s3-devkitc-1 dependencies...\nTool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3\n\n\nUnknownPackageError: Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3' requirements for your system 'darwin_x86_64'"

Erpino7, 2024-07-08:
gh repo clone Erpino7/edoerpani

le-wei, 2024-08-18:
PIO Core Call Error: "The following files/directories have been created in C:\\Users\\LW\\Documents\\PlatformIO\\Projects\\demo2\r\ninclude - Put project header files here\r\nlib - Put project specific (private) libraries here\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\nResolving esp32-c3-devkitm-1 dependencies...\r\nTool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3\r\n\n\nUnknownPackageError: Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3' requirements for your system 'windows_amd64'"

RoMa1961, 2024-09-26:
Could not install platform
PIO Core Call Error: "Platform Manager: espressif32@6.8.1 is already installed\r\nTool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5\r\n\n\nUnknownPackageError: Could not find the package with 'espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5' requirements for your system 'windows_arm64'"

xristocodos, 2024-10-23:
using VSCode+PlatformIO extension on Ubuntu 24.04 on Raspberry Pi 5, 
 (this DOES WORK for ESP32 on ESP-IDF plugin, doesn't work for STM32 due to #problem, seen below w/ additional note)

----rpi pico board, via arduino framework, on PlatformIO----
    Resolving pico dependencies...
    Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.90201.0
    UnknownPackageError: Could not find the package with 'platformio/toolchain-gccarmnoneeabi @ ~1.90201.0' requirements for your system 'linux_aarch64'"

----ST Nucleo F446RE board, via STM32Cube framework, on PlatformIO----
   Resolving nucleo_f446re dependencies...
   Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ >=1.60301.0,<1.80000.0
   UnknownPackageError: Could not find the package with 'platformio/toolchain-gccarmnoneeabi @ >=1.60301.0,<1.80000.0' requirements for your system 'linux_aarch64'"

now, in ubuntu i've installed the toolchain-gccarmnoneeabi  ...-related packages, hoping it would help even though I'm pretty sure PlatformIO prefers to pull down its own separate copy? Idk, I just really want to use my Raspi5 as a dev setup for embedded and this seems like a trivial mismatch somewhere since the gcc-arm-none-eabi packages definitely exist for aarch64. I ran:
$ sudo apt install gcc-arm-none-eabi libnewlib-arm-none-eabi libstdc++-arm-none-eabi-newlib
and installed them + associated dev dependencies. still didn't help.

What can I do or what code can I submit a pull request to, to get PlatformIO to pull down the correct gcc-arm-none-eabi for linux_aarch64? Ubuntu is able to do it on its own, so wheres the disconnect?

(Note: Using the ST Nucleo F446RE with Arduino framework oddly did work, but not for the frameworks Zephyr RTOS, mbed, or STMCube32)

Please let me know if I'm missing any valuable info and I'll handle it.

Dale3691, 2024-10-23:
Pieace of fuckin shit. software

ARLunan, 2024-11-26:
Install Raspberry Pi Pico on VSCode PlatformIO on a MAC Mi Virtual Machine Linux 22.04 (aarch/arm64) . Successfully install PlatformIO. Create a New Project: Raspberry Pi Pico, Select a Board: Raspberry Pi Pico, Framewordk: Arduino . Error Message Could not find the package with 'platformio/toolchain-gccarmnoneeabi@-1.90201.0' requiremenst for your system "linux_aarch64"

edo44, 2024-12-09:
PIO Core Call Error: "The following files/directories have been created in C:\\Users\\ed\\Documents\\PlatformIO\\Projects\r\ninclude - Put project header files here\r\nlib - Put project specific (private) libraries here\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\nResolving lolin_s2_mini dependencies...\r\nTool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3\r\n\n\nUnknownPackageError: Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3' requirements for your system 'windows_amd64'"