platformio/platformio-home #2163: Error occurred while glob [{"name":"HDC2010 Temperature Read","base":"examples/HDC2010_Temperature","files":["HDC2010_Temperature.ino"]},{"name":"HDC2010 Humidity Read","base":"examples/HDC2010_Humidity","files":["HDC2010_Humidity.ino"]},{"name":"MPL3115A2 Pressure Read","base":"examples/MPL3115A2_Pressure","files":["MPL3115A2_Pressure.ino"]},{"name":"SHT21 Humidity Read","base":"examples/SHT21_Humidity","files":["SHT21_Humidity.ino"]},{"name":"SHT21 Temperature Read","base":"examples/SHT21_Temperature","files":["SHT21_Temperature.ino"]},{"name":"TSL2561 Light Read","base":"examples/TSL2561_Light","files":["TSL2561_Light.ino"]}]

Flotilla42, 2021-01-23:
Seems like a rather cryptic description. Was attempting to look at akkoyun/Environment included examples, didn't go too well. 

Server error: {"type":"TypeError","args":["join() argument must be str, bytes, or os.PathLike object, not 'dict'"],"message":"join() argument must be str, bytes, or os.PathLike object, not 'dict'"}

aemondis, 2021-01-26:
Same issue for me, but with a different library - the BSEC Software Library (v1.6.1480). No idea what it means, it looks like a parser error of sorts, possibly due to the examples being in .INO format?

Error occurred while glob [{"name":"Basic usage","base":"examples/basic","files":["basic.ino"]},{"name":"Basic usage with config state saves","base":"examples/basic_config_state","files":["basic_config_state.ino"]},{"name":"Basic usage with config state saves and multiple sensors","base":"examples/basic_config_state_multi","files":["basic_config_state_multi.ino"]},{"name":"Basic usage with config state saves and triggered ULP plus measurements","base":"examples/basic_config_state_ulp_plus","files":["basic_config_state_ulp_plus.ino"]},{"name":"ESP32 deep sleep","base":"examples/esp32DeepSleep","files":["esp32DeepSleep.ino"]}]

Server error: {"type":"TypeError","args":["join() argument must be str or bytes, not 'dict'"],"message":"join() argument must be str or bytes, not 'dict'"}

ivankravets, 2021-01-26:
Fixed in https://github.com/platformio/platformio-home/commit/5f878104161b3add6a70e9c2621d25fa0415849f

The new release of PIO Home is coming soon.

closed 2021-01-26 by ivankravets