platformio/platformio-home #2191: Could not initialize project

balsub123, 2021-01-27:
PIO Core Call Error: "The current working directory C:\\Users\\Balasubramaniam\\Documents\\PlatformIO\\Projects\\Test_App will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\Balasubramaniam\\Documents\\PlatformIO\\Projects\\Test_App\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing ttgo-t-watch (platform: espressif32; board: ttgo-t-watch; framework: espidf)\r\n--------------------------------------------------------------------------------\r\nTool Manager: Installing platformio/toolchain-xtensa32 @ ~2.80200.0\r\nTool Manager: toolchain-xtensa32 @ 2.80200.200827 has been installed!\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-t-watch.html\r\nPLATFORM: Espressif 32 (2.1.0) > TTGO T-Watch\r\nHARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash\r\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\r\nPACKAGES: \r\n - framework-espidf 3.40100.200827 (4.1.0) \r\n - tool-cmake 3.16.4 \r\n - tool-esptoolpy 1.30000.201119 (3.0.0) \r\n - tool-idf 1.0.1 \r\n - tool-mconf 1.4060000.20190628 (406.0.0) \r\n - tool-ninja 1.9.0 \r\n - toolchain-esp32ulp 1.22851.191205 (2.28.51) \r\n - toolchain-xtensa32 2.80200.200827 (8.2.0)\r\nReading CMake configuration...\r\nAssertionError: :\r\n  File \"C:\\Users\\Balasubramaniam\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\main.py\", line 170:\r\n    env.SConscript(\"$BUILD_SCRIPT\")\r\n  File \"C:\\Users\\Balasubramaniam\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Script\\SConscript.py\", line 598:\r\n    return _SConscript(self.fs, *files, **subst_kw)\r\n  File \"C:\\Users\\Balasubramaniam\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Script\\SConscript.py\", line 287:\r\n    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\r\n  File \"C:\\Users\\Balasubramaniam\\.platformio\\platforms\\espressif32\\builder\\main.py\", line 224:\r\n    target_elf = env.BuildProgram()\r\n  File \"C:\\Users\\Balasubramaniam\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Environment.py\", line 219:\r\n    return self.method(*nargs, **kwargs)\r\n  File \"C:\\Users\\Balasubramaniam\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 61:\r\n    env.ProcessProgramDeps()\r\n  File \"C:\\Users\\Balasubramaniam\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Environment.py\", line 219:\r\n    return self.method(*nargs, **kwargs)\r\n  File \"C:\\Users\\Balasubramaniam\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 127:\r\n    env.BuildFrameworks(env.get(\"PIOFRAMEWORK\"))\r\n  File \"C:\\Users\\Balasubramaniam\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Environment.py\", line 219:\r\n    return self.method(*nargs, **kwargs)\r\n  File \"C:\\Users\\Balasubramaniam\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 342:\r\n    SConscript(env.GetFrameworkScript(f), exports=\"env\")\r\n  File \"C:\\Users\\Balasubramaniam\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Script\\SConscript.py\", line 661:\r\n    return method(*args, **kw)\r\n  File \"C:\\Users\\Balasubramaniam\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Script\\SConscript.py\", line 598:\r\n    return _SConscript(self.fs, *files, **subst_kw)\r\n  File \"C:\\Users\\Balasubramaniam\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Script\\SConscript.py\", line 287:\r\n    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\r\n  File \"C:\\Users\\Balasubramaniam\\.platformio\\platforms\\espressif32\\builder\\frameworks\\espidf.py\", line 1180:\r\n    {\"name\": \"ota\", \"type\": \"data\", \"subtype\": \"ota\"},\r\n  File \"C:\\Users\\Balasubramaniam\\.platformio\\platforms\\espressif32\\builder\\frameworks\\espidf.py\", line 838:\r\n    assert os.path.isfile(pt_path)\r\n========================= [FAILED] Took 58.02 seconds ========================="

fancellu, 2021-01-30:
Me too. Latest upgrade appears to have broken many many things.

ivankravets, 2021-02-05:
Fixed in the latest dev-platform. Please update PlatformIO Core and dev-platforms to the latest versions.

closed 2021-02-05 by ivankravets

balsub123, 2021-02-06:
I'll try. Thanks for your support.