platformio/platformio-home #2198: Could not initialize project

bmanfoui, 2021-01-29:
PIO Core Call Error: "The current working directory C:\\Users\\Bryan\\Documents\\PlatformIO\\Projects\\Learning Arduino will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\Bryan\\Documents\\PlatformIO\\Projects\\Learning Arduino\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing metro (platform: atmelavr; board: metro; framework: arduino)\r\n--------------------------------------------------------------------------------\r\nTool Manager: Installing platformio/tool-scons @ ~4.40100.2\r\nTool Manager: tool-scons @ 4.40100.2 has been installed!\r\nTool Manager: Removing tool-scons @ 2.20501.4\r\nTool Manager: tool-scons @ 2.20501.4 has been removed!\r\nVerbose mode can be enabled via `-v, --verbose` option\r\n  File \"C:\\Users\\Bryan\\.platformio\\platforms\\atmelavr\\builder\\main.py\", line 30\r\n\r\n    print \"Please unplug/plug device ...\"\r\n\r\n                                        ^\r\n\r\nSyntaxError: Missing parentheses in call to 'print'. Did you mean print(\"Please unplug/plug device ...\")?\r\n\r\n========================== [FAILED] Took 9.55 seconds =========================="

ivankravets, 2021-01-29:
Please remove `C:\Users\Bryan\.platformio\penv` folder and restart VSCode.

closed 2021-01-29 by ivankravets

bmanfoui, 2021-01-30:
Thank you

That worked

From: Ivan Kravets
Sent: January 29, 2021 3:01 PM
To: platformio/platformio-home
Cc: bmanfoui; Author
Subject: Re: [platformio/platformio-home] Could not initialize project (#2198)

Please remove C:\Users\Bryan\.platformio\penv folder and restart VSCode.