platformio/platformio-home #2316: Windows WSL: I can't open any files(projects) in the PIO open browser.

Kamilake, 2021-03-03:
### Configuration

**Operating system**:
Windows 10 20H2 with WSL Ubuntu 20.10 (Groovy Gorilla)

VScode 1.53.2

**PlatformIO Version** (`platformio --version`):

PlatformIO Core, version 5.1.0

### Description of problem

#### Steps to Reproduce

1. Press button Make Project or open Project
2. 
[image]

### Actual Results
No Items is displayed in most directories.
(I guess it's probably a permission issue)
[image]

### Expected Results

[image]
I should be able to open the project in the path I want.

### If problems with PlatformIO Build System:

### Additional info

I will provide any additional information needed.
I apologize for the lack of information.

unsplorer, 2021-10-05:
I've managed to hack together a possible solution by changing Ln 19 in app/modules/core/paths.js
This is obviously very poor practice, but if you're only using Platformio by remoting in from a windows host to WSL or a Unix host then it may work. 
[image]

Kamilake, 2021-10-05:
> I've managed to hack together a possible solution by changing Ln 19 in app/modules/core/paths.js This is obviously very poor practice, but if you're only using Platformio by remoting in from a windows host to WSL or a Unix host then it may work. [image]

I couldn't figure out the path to that, but I just checked. It worked for me :)

chrishdx, 2021-11-19:
> I've managed to hack together a possible solution by changing Ln 19 in app/modules/core/paths.js This is obviously very poor practice, but if you're only using Platformio by remoting in from a windows host to WSL or a Unix host then it may work. [image]

Hi,

how did you do that when i install via pip3 platformio i can't find the file anywhere.

cbundy, 2022-03-18:
When the app bootstraps initially, it gets the platform architecture from the host - e.g. `coreSystype: "linux_aarch64"` - could the front end use the host's OS/architecture instead of that of the browsers?

Squirrelcoding, 2022-05-31:
I know this is like a year old but how exactly do I solve this?

Kamilake, 2022-05-31:
> I know this is like a year old but how exactly do I solve this?

Maybe I modified the plug-in path directly at the time..
I don't remember well because it's been a while, but is the problem still there?

Squirrelcoding, 2022-05-31:
> Maybe I modified the plug-in path directly at the time.. I don't remember well because it's been a while, but is the problem still there?

Yeah, I saw that the solution involved editing the source code and stuff, how exactly did you guys edit the source code?

unsplorer, 2022-06-09:
So ive long forgotten how I originally did this, but I came back to it and ended up editing the file at /home/$USER/.platformio/packages/contrib-piohome/main.*********.min.js . I then searched the code for "\\":"/" and replaced that with "/":"/" . Very unorthodox, but it worked for me this time around.

hamed3d, 2022-07-15:
> So ive long forgotten how I originally did this, but I came back to it and ended up editing the file at /home/$USER/.platformio/packages/contrib-piohome/main.*********.min.js . I then searched the [...]

@unsplorer, you are a genius! This solved my problem after hours of trying to find the reason for the "Invalid Params" issue I was receiving on WSL 2 with a Devcontainer setup. As a temporary fix, I created a bash command to automate the steps post-installation.

`#!/bin/bash
cd /home/$USER/.platformio/packages/contrib-piohome/
find . -maxdepth 1 -name "main*" -exec sed -i 's#"\\\\":"/"#"/":"/"#g' {} +`

This should be added as a setting for PlatformIO IDE so that users can decide to use \\ on windows or / on Linux

anisoptera, 2022-08-14:
Just encountered this issue, and the posted fixes do not work for me. How is it so difficult to solve this issue? The program is running on a Linux system, what possible reason could the program have to use Windows path separators? Why would you translate separators client-side based on the client browser OS instead of the server process??

the-moog, 2022-08-18:
VSCode in Win10 Remote -> WSL
Windows 10 understands Posix path separators so this is not a Windows bug, it is a bug in internal path parsing in PlatformIO. Opening the same folder in VSCode (not in PlatformIO inside VSCode) opens the folder correctly.

If you create a new project 'Test1' in PlatformIO under WSL it creates a folder called 'Projects/Test1' instead of two folders, one  called 'Projects' that contains a folder called 'Test1'.  Note the slash direction.

If you try and correct that and make a folder called Projects containing Test1 and copy the above project into it and then try and 'Add existing' it says the folder 'Test1' is empty.

Conclusion file open and file save are both badly broken.

Proof Windows 10 understands Poxix path separators.
START -> RUN -> CMD
```cmd
U:\folder> c:
C:\home> cd "/Windows/System32"
C:\Windows\System32>
```

Kamilake, 2022-08-18:
> VSCode in Win10 Remote -> WSL Windows 10 understands Posix path separators so this is not a Windows bug, it is a bug in internal path parsing in PlatformIO. Opening the same folder in VSCode (not in PlatformIO inside VSCode) opens the folder correctly.

I have no idea why PIO doesn't use POSIX style paths.

delhub, 2022-10-15:
Has anyone come up with a workaround or solution to this issue?

cbundy, 2022-10-22:
I had a fork of this working previously, but I never got enough testing done to create a PR. 

Essentially, the platform we are detecting needs to be that where PIO is running, not the front end (in my case, linux via SSH for PIO and VS Code on Windows for the front end)

I will try and get it setup again, though I'm not particularly familiar with what testing we'd require maybe it would be a start

cbundy, 2022-10-23:
I've created a PR here: https://github.com/platformio/platformio-home/pull/5018

paule96, 2023-03-05:
can somebody revisit the PR? It's a really annoying issue that makes it quite hard to use dev containers with platform io

Sammy1Am, 2023-03-06:
+1

Can't even get PlatformIO to open existing projects because it can't figure out how to navigate the folder structure.

Kamilake, 2023-03-23:
.....

goesbysteve, 2023-04-23:
Thank you all who've contributed to this. Made my day a lot easier.

italocjs, 2024-04-30:
Tip: i found that i get a LOT more performance by installing platformio and cloning the project inside wsl, then connecting to it remotely.

I5 10300H Windows 11 ~10min non chached  | 4min cached
I5 10300H Win 11 WSL ~ 3min non cached | 7s cached

gpmontt, 2025-02-05:
I saw the same performance using WSL and PlatformIO. Maybe is possible to rethink the way, how use the '/'  for Unix and  '\\' for windows

cbundy, 2025-02-12:
➡ Please see the owner of the repository noting that [it will be revisited in the v7 release](https://github.com/platformio/platformio-home/pull/5018#issuecomment-1937531447), but **it will not** be considered as a fix in the current version

ℹ The best thing you could do for now is to pull the source and cherry pick the changes into your own custom version.

dejwk, 2025-12-30:
I can confirm that the editing the file ~/.platformio/packages/contrib-piohome/main.min.js still works, only you need to replace "\\" with "\" (there are 3 occurences in core version 6.1.18).