platformio/platformio-home #2636: Could not initialize project

UnlimitedKaos, 2021-06-03:
Hello, I am trying to create a bluepill project, I can create projects for arduinos and esp32. I dont know how to fix this

```
PIO Core Call Error:  "The current working directory C:\\Users\\chaos\\Documents\\PlatformIO\\Projects\\new_stm32_test will be used for the project.

The next files/directories have been created in C:\\Users\\chaos\\Documents\\PlatformIO\\Projects\\new_stm32_test\r\ninclude - Put project header files here
lib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File
Platform Manager: Installing ststm32\r\nPlatform Manager: ststm32 @ 14.0.0 has been installed!
Tool Manager: toolchain-gccarmnoneeabi @ 1.70201.0 is already installed
The platform 'ststm32' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.

Error: Processing bluepill_f103c8 (platform: ststm32; board: bluepill_f103c8; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
PLATFORM: ST STM32 (14.0.0) > BluePill F103C8
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (stlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.20000.210601 (2.0.0) 
 - framework-cmsis 2.50700.210515 (5.7.0) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
ImportError: cannot import name 'IS_WINDOWS' from 'platformio.compat' (c:\\users\\chaos\\.platformio\\penv\\lib\\site-packages\\platformio\\compat.py):
 File \"C:\\Users\\chaos\\.platformio\\penv\\Lib\\site-packages\\platformio\\builder\\main.py\", line 177:\r\n env.SConscript(\"$BUILD_SCRIPT\")
 File \"C:\\Users\\chaos\\.platformio\\packages\\tool-scons\\scons-local-4.1.0\\SCons\\Script\\SConscript.py\", line 591:
 return _SConscript(self.fs, *files, **subst_kw)
 File \"C:\\Users\\chaos\\.platformio\\packages\\tool-scons\\scons-local-4.1.0\\SCons\\Script\\SConscript.py\", line 280:
 exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
 File \"C:\\Users\\chaos\\.platformio\\platforms\\ststm32\\builder\\main.py\", line 120:
 target_elf = env.BuildProgram()
 File \"C:\\Users\\chaos\\.platformio\\packages\\tool-scons\\scons-local-4.1.0\\SCons\\Util.py\", line 658:
 return self.method(*nargs, **kwargs)\r\n File \"C:\\Users\\chaos\\.platformio\\penv\\Lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 61:
 env.ProcessProgramDeps()\r\n File \"C:\\Users\\chaos\\.platformio\\packages\\tool-scons\\scons-local-4.1.0\\SCons\\Util.py\", line 658:
 return self.method(*nargs, **kwargs)\r\n File \"C:\\Users\\chaos\\.platformio\\penv\\Lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 127:
 env.BuildFrameworks(env.get(\"PIOFRAMEWORK\"))
 File \"C:\\Users\\chaos\\.platformio\\packages\\tool-scons\\scons-local-4.1.0\\SCons\\Util.py\", line 658:
 return self.method(*nargs, **kwargs)\r\n File \"C:\\Users\\chaos\\.platformio\\penv\\Lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 342:
 SConscript(env.GetFrameworkScript(f), exports=\"env\")
 File \"C:\\Users\\chaos\\.platformio\\packages\\tool-scons\\scons-local-4.1.0\\SCons\\Script\\SConscript.py\", line 654:
 return method(*args, **kw)
 File \"C:\\Users\\chaos\\.platformio\\packages\\tool-scons\\scons-local-4.1.0\\SCons\\Script\\SConscript.py\", line 591:
 return _SConscript(self.fs, *files, **subst_kw)\r\n File \"C:\\Users\\chaos\\.platformio\\packages\\tool-scons\\scons-local-4.1.0\\SCons\\Script\\SConscript.py\", line 280:
 exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\r\n File \"C:\\Users\\chaos\\.platformio\\platforms\\ststm32\\builder\\frameworks\\arduino.py\", line 50:
 SConscript(build_script)
 File \"C:\\Users\\chaos\\.platformio\\packages\\tool-scons\\scons-local-4.1.0\\SCons\\Script\\SConscript.py\", line 654:
 return method(*args, **kw)
 File \"C:\\Users\\chaos\\.platformio\\packages\\tool-scons\\scons-local-4.1.0\\SCons\\Script\\SConscript.py\", line 591:
 return _SConscript(self.fs, *files, **subst_kw)
 File \"C:\\Users\\chaos\\.platformio\\packages\\tool-scons\\scons-local-4.1.0\\SCons\\Script\\SConscript.py\", line 280:
 exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
 File \"C:\\Users\\chaos\\.platformio\\packages\\framework-arduinoststm32\\tools\\platformio\\platformio-build.py\", line 28:
 from platformio.compat import IS_WINDOWS
 ========================== [FAILED] Took 0.74 seconds =========================="

```

UnlimitedKaos, 2021-06-03:
I fixed it by going into `C:\users\{user}\.platformio\penv\lib\site-packages\platformio\compat.py` and adding `IS_WINDOWS = WINDOWS` under the `sys.platform.startswith()` lines.

Im not sure if this is the proper way to do things, but I can create projects now

RA2lover, 2021-06-04:
#2635, #2634, #2632, #2630 and #2629 and platformio/platform-ststm32#529 all show the same error.

I was able to get the fix to work on my environment, though due to some other troubleshooting procedures taken earlier it was located at `C:\.platformio` instead of `C:\users\{user}\.platformio`.

All reported issues tried to use the stm32 platform using the arduino-stm32 framework, and i think a proper fix would be better done there.

UnlimitedKaos, 2021-06-04:
Makes sense, I went to `.platformio\packages\framework-arduinoststm32\tools\platformio\platformio-build.py` and changed `IS_WINDOWS` to `WINDOWS` in line 28 and 411, as well as undoing my previous change. 

Creating projects with the arduino framework works fine now

closed 2021-06-04 by UnlimitedKaos

RA2lover, 2021-06-04:
running `pio update`, turns out my environment had been running `framework-arduinoststm32  4.20000.210601` instead of the current latest version, `framework-arduinoststm32  4.20000.210602`, which can create projects properly despite removing the fix.

UnlimitedKaos, 2021-06-04:
oh, lol, I guess I shouldve checked for updates first