platformio/platformio-home #2923: Could not initialize project

UsagiB4, 2021-08-08:
`Invalid params`
Can not create new project.
I've tried reinstalling **PlatformIO** ( Clearing all data ). Still the same.
It was working fine and I was able to create new projects and upload them.
But today, first I got error while uploading the code, and now, I can not create new project........

UsagiB4, 2021-08-10:
### SOLVED 
my anti-virus was blocking **VSC** to create new file. 
Just added **VSC** to white list and it's working fine.....

closed 2021-08-10 by UsagiB4

Kinetic69, 2024-05-24:
Go to Start / Windows Security / Virus & threat protection / Manage ransomware production (at the bottom) / Block history, expand it / <Yes>. You'll find python.exe blocked by controlled folder access. 

Allow it.

Create a new project, it should work.  If not,  and see what other processes are getting blocked.

[image]

[image]

GabTibolla, 2024-06-06:
show de bola marcio. Funcionou

awolhj, 2025-04-08:
> Go to Start / Windows Security / Virus & threat protection / Manage ransomware production (at the bottom) / Block history, expand it / . You'll find python.exe blocked by controlled folder access.
> 
> Allow it.
> 
> Create a new project, it should work. If not, and see what other processes are getting blocked.
> 
> <img alt="image" width="391" src="https://private-user-images.githubusercontent.com/83773346/333602392-0a3182a7-fe66-48dc-899b-00d108a09053.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDQwNzM4MDEsIm5iZiI6MTc0NDA3MzUwMSwicGF0aCI6Ii84Mzc3MzM0Ni8zMzM2MDIzOTItMGEzMTgyYTctZmU2Ni00OGRjLTg5OWItMDBkMTA4YTA5MDUzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDA4VDAwNTE0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVkYTJkZDlmODM5OWM3NmIwZmFmMjUzZTA0ZWNkY2RiMTY2OThjN2IxYTVjMzE4YWQwMDNjODZjYTAzY2NkMTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.lq43Pu22Y0wFukxtnbhkAcDfr6iDHIfuvvM_yqtUsnQ"> <img alt="image" width="344" src="https://private-user-images.githubusercontent.com/83773346/333602602-b51b4efb-62d9-4f43-b397-49e1bb8924a6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDQwNzM4MDEsIm5iZiI6MTc0NDA3MzUwMSwicGF0aCI6Ii84Mzc3MzM0Ni8zMzM2MDI2MDItYjUxYjRlZmItNjJkOS00ZjQzLWIzOTctNDllMWJiODkyNGE2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDA4VDAwNTE0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWViZWMzOGIyZGIyMWUxZWFhMDQ5NjhjODE5MDk2YmNlNWQ0NDBlODY4ZGQxMjhhOTJmNTZhZTRiYzhmMmJiNWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.598zwIC3lUI3_yyxW9tXq9aXBFXboNPQTJTLouNyp5Q">

Your very effective answer!