platformio/platformio-home #3316: MBED: cannot import name 'Iterable' from 'collections'

ase1590, 2021-11-15:
This looks to be stemming from pygal, which deprecated the 'collections' function in python 3.9. 
Now that 3.10 is out, this seems to be broken. 

https://github.com/Kozea/pygal/issues/500

PIO Core Call Error: "The current working directory /home/ase1590/Documents/PlatformIO/Projects/test will be used for the project.\n\nThe next files/directories have been created in /home/ase1590/Documents/PlatformIO/Projects/test\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing lpc1768 (platform: nxplpc; board: lpc1768; framework: mbed)\n--------------------------------------------------------------------------------\nTool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.90201.0\nDownloading\nUnpacking\nTool Manager: toolchain-gccarmnoneeabi @ 1.90201.191206 has been installed!\nTool Manager: Installing platformio/framework-mbed @ ~6.60900.0\nDownloading\nUnpacking\nTool Manager: framework-mbed @ 6.60900.210318 has been installed!\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/nxplpc/lpc1768.html\nPLATFORM: NXP LPC (9.0.0) > NXP mbed LPC1768\nHARDWARE: LPC1768 96MHz, 64KB RAM, 512KB Flash\nDEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)\nPACKAGES: \n - framework-mbed 6.60900.210318 (6.9.0) \n - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)\n**ImportError: cannot import name 'Iterable' from 'collections' (/usr/lib64/python3.10/collections/__init__.py):\n**  File \"/home/ase1590/.platformio/penv/lib/python3.10/site-packages/platformio/builder/main.py\", line 180:\n    env.SConscript(\"$BUILD_SCRIPT\")\n  File \"/home/ase1590/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Script/SConscript.py\", line 597:\n    return _SConscript(self.fs, *files, **subst_kw)\n  File \"/home/ase1590/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Script/SConscript.py\", line 285:\n    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\n  File \"/home/ase1590/.platformio/platforms/nxplpc/builder/main.py\", line 117:\n    target_elf = env.BuildProgram()\n  File \"/home/ase1590/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Util.py\", line 748:\n    return self.method(*nargs, **kwargs)\n  File \"/home/ase1590/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/platformio.py\", line 61:\n    env.ProcessProgramDeps()\n  File \"/home/ase1590/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Util.py\", line 748:\n    return self.method(*nargs, **kwargs)\n  File \"/home/ase1590/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/platformio.py\", line 128:\n    env.BuildFrameworks(env.get(\"PIOFRAMEWORK\"))\n  File \"/home/ase1590/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Util.py\", line 748:\n    return self.method(*nargs, **kwargs)\n  File \"/home/ase1590/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/platformio.py\", line 343:\n    SConscript(env.GetFrameworkScript(f), exports=\"env\")\n  File \"/home/ase1590/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Script/SConscript.py\", line 660:\n    return method(*args, **kw)\n  File \"/home/ase1590/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Script/SConscript.py\", line 597:\n    return _SConscript(self.fs, *files, **subst_kw)\n  File \"/home/ase1590/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Script/SConscript.py\", line 285:\n    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\n  File \"/home/ase1590/.platformio/platforms/nxplpc/builder/frameworks/mbed.py\", line 35:\n    SConscript(\n  File \"/home/ase1590/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Script/SConscript.py\", line 660:\n    return method(*args, **kw)\n  File \"/home/ase1590/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Script/SConscript.py\", line 597:\n    return _SConscript(self.fs, *files, **subst_kw)\n  File \"/home/ase1590/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Script/SConscript.py\", line 285:\n    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\n  File \"/home/ase1590/.platformio/packages/framework-mbed/platformio/platformio-build.py\", line 44:\n    from pio_mbed_adapter import PlatformioMbedAdapter\n  File \"/home/ase1590/.platformio/packages/framework-mbed/platformio/pio_mbed_adapter.py\", line 23:\n    from tools.build_api import prepare_toolchain, UPDATE_WHITELIST\n  File \"/home/ase1590/.platformio/packages/framework-mbed/tools/build_api.py\", line 48:\n    from .libraries import Library\n  File \"/home/ase1590/.platformio/packages/framework-mbed/tools/libraries.py\", line 25:\n    from tools.tests import TEST_MBED_LIB, DEFAULT_SUPPORT\n  File \"/home/ase1590/.platformio/packages/framework-mbed/tools/tests.py\", line 18:\n    from past.builtins import cmp\n  File \"/home/ase1590/.platformio/packages/framework-mbed/platformio/package_deps/py3/past/builtins/__init__.py\", line 43:\n    from past.builtins.noniterators import (filter, map, range, reduce, zip)\n  File \"/home/ase1590/.platformio/packages/framework-mbed/platformio/package_deps/py3/past/builtins/noniterators.py\", line 24:\n    from past.types import basestring\n  File \"/home/ase1590/.platformio/packages/framework-mbed/platformio/package_deps/py3/past/types/__init__.py\", line 25:\n    from .oldstr import oldstr\n  File \"/home/ase1590/.platformio/packages/framework-mbed/platformio/package_deps/py3/past/types/oldstr.py\", line 5:\n    from collections import Iterable\n========================= [FAILED] Took 107.50 seconds ========================="

closed 2021-12-23 by stale[bot]

ducky64, 2022-01-20:
Error seems related to non-backwards-compatible naming changes in Python3.10 and still reproduces. Using Python3.9 does not give this error. This seems to be an issue in a dependency, specifically oldstr.py in past. In fact, it looks like it's pulling in an [absolutely ancient version of oldstr.py](https://github.com/PythonCharmers/python-future/commit/02fdb329d782a1bbf6c08700d63e95a5de01fd86).

Related discussion in a different project here: https://github.com/Azure/azure-cli/issues/20444

Here is a change in a different project that fixes this issue: https://github.com/Kozea/pygal/issues/514

The [latest version of oldstr.py](https://github.com/PythonCharmers/python-future/blob/master/src/past/types/oldstr.py) might also fix this issue.

asmello, 2022-08-22:
Can we reopen? This is still an issue.

Ling-Jun, 2022-09-02:
Still an issue for me, here is my stack trace. 

2022-09-02T18:13:04.6202509Z ##[section]Starting: Train model
2022-09-02T18:13:04.6253055Z ==============================================================================
2022-09-02T18:13:04.6253390Z Task         : Azure CLI
2022-09-02T18:13:04.6253865Z Description  : Run Azure CLI commands against an Azure subscription in a PowerShell Core/Shell script when running on Linux agent or PowerShell/PowerShell Core/Batch script when running on Windows agent.
2022-09-02T18:13:04.6254343Z Version      : 2.198.0
2022-09-02T18:13:04.6254560Z Author       : Microsoft Corporation
2022-09-02T18:13:04.6254899Z Help         : https://docs.microsoft.com/azure/devops/pipelines/tasks/deploy/azure-cli
2022-09-02T18:13:04.6255297Z ==============================================================================
2022-09-02T18:13:04.7599192Z [command]/usr/bin/az --version
2022-09-02T18:13:05.2640315Z WARNING: You have 3 updates available. Consider updating your CLI installation with 'az upgrade'
2022-09-02T18:13:05.2640683Z 
2022-09-02T18:13:05.2641061Z Please let us know how we are doing: https://aka.ms/azureclihats
2022-09-02T18:13:05.2641959Z and let us know if you're interested in trying out our newest features: https://aka.ms/CLIUXstudy
2022-09-02T18:13:05.2642609Z azure-cli                         2.38.0 *
2022-09-02T18:13:05.2642762Z 
2022-09-02T18:13:05.2643044Z core                              2.38.0 *
2022-09-02T18:13:05.2643361Z telemetry                          1.0.6 *
2022-09-02T18:13:05.2643523Z 
2022-09-02T18:13:05.2643768Z Extensions:
2022-09-02T18:13:05.2644236Z azure-devops                      0.25.0
2022-09-02T18:13:05.2644734Z azure-cli-ml                      1.41.0
2022-09-02T18:13:05.2644905Z 
2022-09-02T18:13:05.2645155Z Dependencies:
2022-09-02T18:13:05.2645461Z msal                            1.18.0b1
2022-09-02T18:13:05.2645961Z azure-mgmt-resource             21.1.0b1
2022-09-02T18:13:05.2646126Z 
2022-09-02T18:13:05.2646583Z Python location '/opt/az/bin/python3'
2022-09-02T18:13:05.2647132Z Extensions directory '/opt/az/azcliextensions'
2022-09-02T18:13:05.2647344Z 
2022-09-02T18:13:05.2647731Z Python (Linux) 3.10.5 (main, Jul  1 2022, 04:14:49) [GCC 7.5.0]
2022-09-02T18:13:05.2647995Z 
2022-09-02T18:13:05.2648310Z Legal docs and information: aka.ms/AzureCliLegal
2022-09-02T18:13:05.2648503Z 
2022-09-02T18:13:05.2648611Z 
2022-09-02T18:13:05.2656647Z Setting AZURE_CONFIG_DIR env variable to: /home/vsts/work/_temp/.azclitask
2022-09-02T18:13:05.2661788Z Setting active cloud to: AzureCloud
2022-09-02T18:13:05.2723483Z [command]/usr/bin/az cloud set -n AzureCloud
2022-09-02T18:13:05.4889036Z [command]/usr/bin/az login --service-principal -u *** --password=*** --tenant 05c95b33-90ca-49d5-b644-288b930b912b --allow-no-subscriptions
2022-09-02T18:13:06.0465784Z [
2022-09-02T18:13:06.0466718Z   {
2022-09-02T18:13:06.0467602Z     "cloudName": "AzureCloud",
2022-09-02T18:13:06.0468935Z     "homeTenantId": "05c95b33-90ca-49d5-b644-288b930b912b",
2022-09-02T18:13:06.0469784Z     "id": "9985a892-96f1-4cc2-a286-425c8fcea097",
2022-09-02T18:13:06.0470367Z     "isDefault": true,
2022-09-02T18:13:06.0470799Z     "managedByTenants": [
2022-09-02T18:13:06.0471222Z       {
2022-09-02T18:13:06.0471914Z         "tenantId": "2f4a9838-26b7-47ee-be60-ccc1fdec5953"
2022-09-02T18:13:06.0472420Z       }
2022-09-02T18:13:06.0472788Z     ],
2022-09-02T18:13:06.0473226Z     "name": "NRCan Sandbox",
2022-09-02T18:13:06.0473681Z     "state": "Enabled",
2022-09-02T18:13:06.0474374Z     "tenantId": "05c95b33-90ca-49d5-b644-288b930b912b",
2022-09-02T18:13:06.0474890Z     "user": {
2022-09-02T18:13:06.0476218Z       "name": "***",
2022-09-02T18:13:06.0476726Z       "type": "servicePrincipal"
2022-09-02T18:13:06.0477152Z     }
2022-09-02T18:13:06.0477523Z   }
2022-09-02T18:13:06.0477894Z ]
2022-09-02T18:13:06.0488220Z [command]/usr/bin/az account set --subscription 9985a892-96f1-4cc2-a286-425c8fcea097
2022-09-02T18:13:06.2720789Z [command]/bin/bash /home/vsts/work/_temp/azureclitaskscript1662142384752.sh
2022-09-02T18:13:11.4340812Z RunId: wksp-expr_1662142388_2c04a6cc
2022-09-02T18:13:11.4342995Z Web View: https://ml.azure.com/runs/wksp-expr_1662142388_2c04a6cc?wsid=/subscriptions/9985a892-96f1-4cc2-a286-425c8fcea097/resourcegroups/rg-nrcan-cmss-digital-accelerator-team-sandbox/workspaces/wksp_name&tid=05c95b33-90ca-49d5-b644-288b930b912b
2022-09-02T18:13:14.0307563Z 
2022-09-02T18:13:14.0309852Z Streaming azureml-logs/20_image_build_log.txt
2022-09-02T18:13:14.0310419Z =============================================
2022-09-02T18:13:14.0310622Z 
2022-09-02T18:13:24.0618225Z 2022/09/02 18:13:15 Downloading source code...
2022-09-02T18:13:24.0618848Z 2022/09/02 18:13:16 Finished downloading source code
[... 939 lines cut ...]
2022-09-02T18:26:19.5969884Z [2022-09-02T18:25:42.032208] job release stage : upload_datastore starting...
2022-09-02T18:26:19.5970876Z [2022-09-02T18:25:42.040232] Entering context manager injector.
2022-09-02T18:26:19.5972303Z [2022-09-02T18:25:42.041533] job release stage : start importing azureml.history._tracking in run_history_release.[2022-09-02T18:25:42.042288] job release stage : copy_batchai_cached_logs starting...
2022-09-02T18:26:19.5974176Z [2022-09-02T18:25:42.042649] job release stage : execute_job_release starting...
2022-09-02T18:26:19.5974696Z 
[... cut ...]
2022-09-02T18:26:19.5989031Z Web View: https://ml.azure.com/runs/wksp-expr_1662142388_2c04a6cc?wsid=/subscriptions/9985a892-96f1-4cc2-a286-425c8fcea097/resourcegroups/rg-nrcan-cmss-digital-accelerator-team-sandbox/workspaces/wksp_name&tid=05c95b33-90ca-49d5-b644-288b930b912b
2022-09-02T18:26:19.5989808Z 
2022-09-02T18:26:20.1677613Z ERROR: cannot import name 'Iterator' from 'collections' (/opt/az/lib/python3.10/collections/__init__.py)
2022-09-02T18:26:20.4007034Z ##[error]Script failed with exit code: 1
2022-09-02T18:26:20.4023666Z [command]/usr/bin/az account clear
2022-09-02T18:26:20.6403648Z ##[section]Finishing: Train model

asmello, 2022-09-02:
Hey, I traced the issue and proposed a fix here: https://github.com/platformio/builder-framework-mbed/pull/31

If we can get that PR merged, it should resolve the issue for everyone.

@Ling-Jun in the meantime you can manually apply that fix to get yourself unblocked, like I did.