platformio/platformio-home #3322: Could not load recent projects

NINO-3D, 2021-11-15:
Server error: {"type":"AssertionError","args":[],"message":""}

alex18881, 2021-11-15:
I'm experiencin the same problm:
Server error: {"type":"JSONDecodeError","args":["Expecting value: line 1 column 1 (char 0)"],"message":"Expecting value: line 1 column 1 (char 0)"}

NINO-3D, 2021-11-15:
I reinstalled the vscode and it seems to have updated and now it works.
The problem lies in platformIO.
I hope I was helpful.

alex18881, 2021-11-15:
I tried to reinstall the platformio. But not the vscode. I'd prefere to update platformio's configuration somehow better than reinstalling abd configuring the whole bunch of extensions I have now.

alex18881, 2021-11-15:
Also when I was googling the possible solution there was a topic on stackoverflow about a similar issue in python. And the solution was not to use littlejson (or something liie that) python library. May be this will help somehow.

closed 2021-12-23 by stale[bot]