platformio/platformio-home #3435: Could not install platform

mshamash, 2021-12-01:
PIO Core Call Error: "Platform Manager: Installing ststm32 @ 12.1.0\n\n\nError: Could not install package 'ststm32 @ 12.1.0' for 'darwin_arm64' system"

This happens on all platforms tested, Windows 10 and 11 (x64), Linux (x64), unsure why

gvvuur3n, 2021-12-01:
Hey, Well I was having the same issue except on a w10 machine and ended up resolving it by turning off my windows firewall, realtime virus & threat protection, tamper protection, cloud protection.. basically everything. 

I also deleted my %userprofile%.platformio\appstate.json for certainty and restarted vscode. 

Doing the above fixed my issue of not being able to install packages when creating a new project. My best guess is that windows is blocking platformio from downloading/installing packages in the background.

Hope this helps.

mshamash, 2021-12-01:
Thanks @gvvuur3n ! That fixed the issue on my Windows 11 install. I disabled Tamper protection, Real-time protection, and Cloud-delivered protection. No need to restart vscode or delete my appstate.json file in this case.

closed 2021-12-01 by mshamash