platformio/platformio-home #3642: Libraries: Updates

PaulSan71, 2022-01-05:
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n  File \"C:\\Users\\pauls\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 121, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"C:\\Users\\pauls\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1128, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"C:\\Users\\pauls\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1053, in main\r\n    rv = self.invoke(ctx)\r\n  File \"C:\\Users\\pauls\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"C:\\Users\\pauls\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1659, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\pauls\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1659, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\pauls\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1395, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"C:\\Users\\pauls\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 754, in invoke\r\n    return __callback(*args, **kwargs)\r\n  File \"C:\\Users\\pauls\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 26, in new_func\r\n    return f(get_current_context(), *args, **kwargs)\r\n  File \"C:\\Users\\pauls\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib\\command.py\", line 262, in lib_update\r\n    outdated = lm.outdated(pkg, spec)\r\n  File \"C:\\Users\\pauls\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\_update.py\", line 49, in outdated\r\n    reg_pkg = self.fetch_registry_package(pkg.metadata.spec)\r\n  File \"C:\\Users\\pauls\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\_registry.py\", line 144, in fetch_registry_package\r\n    result = regclient.get_package(self.pkg_type, spec.owner, spec.name)\r\n  File \"C:\\Users\\pauls\\.platformio\\penv\\lib\\site-packages\\platformio\\clients\\registry.py\", line 125, in get_package\r\n    return self.fetch_json_data(\r\n  File \"C:\\Users\\pauls\\.platformio\\penv\\lib\\site-packages\\platformio\\clients\\http.py\", line 146, in fetch_json_data\r\n    return json.loads(result)\r\n  File \"C:\\Users\\pauls\\.platformio\\python3\\lib\\json\\__init__.py\", line 346, in loads\r\n    return _default_decoder.decode(s)\r\n  File \"C:\\Users\\pauls\\.platformio\\python3\\lib\\json\\decoder.py\", line 337, in decode\r\n    obj, end = self.raw_decode(s, idx=_w(s, 0).end())\r\n  File \"C:\\Users\\pauls\\.platformio\\python3\\lib\\json\\decoder.py\", line 355, in raw_decode\r\n    raise JSONDecodeError(\"Expecting value\", s, err.value) from None\r\njson.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

PaulSan71, 2022-01-05:
This was automatically generated by PlatformIo when I clicked on Libraries/Update

shumtravi, 2022-01-06:
Same issue

PaulSan71, 2022-01-06:
I was finally able to get Update to work without generating the above error message.  Once all of my libraries were updated, everything worked correctly.. For me, this is a non-issue, now.

closed 2022-02-12 by stale[bot]