platformio/platformio-home #4007: Could not load mDNS services

siv237, 2022-03-15:
PIO Core Call Error: "Error: Traceback (most recent call last):\n  File \"/root/.platformio/penv/lib/python3.6/site-packages/platformio/__main__.py\", line 121, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"/root/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 1128, in __call__\n    return self.main(*args, **kwargs)\n  File \"/root/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 1053, in main\n    rv = self.invoke(ctx)\n  File \"/root/.platformio/penv/lib/python3.6/site-packages/platformio/commands/__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"/root/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 1659, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/root/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 1659, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/root/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 1395, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"/root/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 754, in invoke\n    return __callback(*args, **kwargs)\n  File \"/root/.platformio/penv/lib/python3.6/site-packages/platformio/commands/device/command.py\", line 50, in device_list\n    data[\"mdns\"] = util.get_mdns_services()\n  File \"/root/.platformio/penv/lib/python3.6/site-packages/platformio/util.py\", line 206, in get_mdns_services\n    with mDNSListener() as mdns:\n  File \"/root/.platformio/penv/lib/python3.6/site-packages/platformio/util.py\", line 163, in __init__\n    self._zc = zeroconf.Zeroconf(interfaces=zeroconf.InterfaceChoice.All)\n  File \"/root/.platformio/penv/lib/python3.6/site-packages/zeroconf/_core.py\", line 467, in __init__\n    self.start()\n  File \"/root/.platformio/penv/lib/python3.6/site-packages/zeroconf/_core.py\", line 471, in start\n    self.loop = get_running_loop()\n  File \"/root/.platformio/penv/lib/python3.6/site-packages/zeroconf/_utils/asyncio.py\", line 121, in get_running_loop\n    return asyncio.get_running_loop()\nAttributeError: module 'asyncio' has no attribute 'get_running_loop'\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

siv237, 2022-03-15:
not work scan mDNS devices

Techi-Dev, 2022-04-02:
Try this solution from stackoverflow 
https://stackoverflow.com/questions/65102892/cannot-create-new-projects-in-platform-io-vs-code-plugin.

closed 2022-10-20 by stale[bot]