platformio/platformio-home #4198: Could not initialize project

xinyew, 2022-04-24:
PIO Core Call Error: "The current working directory /Users/xinye/Desktop/u?ntitled folder/ttt will be used for the project.\n\nThe next files/directories have been created in /Users/xinye/Desktop/u?ntitled folder/ttt\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing particle_argon (platform: nordicnrf52; board: particle_argon; framework: zephyr)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nKeyError: \"Invalid board option 'build.softdevice.sd_fwid'\":\n  File \"/Users/xinye/.platformio/penv/lib/python3.9/site-packages/platformio/builder/main.py\", line 181:\n    env.SConscript(\"$BUILD_SCRIPT\")\n  File \"/Users/xinye/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/SConscript.py\", line 597:\n    return _SConscript(self.fs, *files, **subst_kw)\n  File \"/Users/xinye/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/SConscript.py\", line 285:\n    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\n  File \"/Users/xinye/.platformio/platforms/nordicnrf52/builder/main.py\", line 143:\n    board.get(\"build.softdevice.sd_fwid\"),\n  File \"/Users/xinye/.platformio/penv/lib/python3.9/site-packages/platformio/platform/board.py\", line 46:\n    raise KeyError(\"Invalid board option '%s'\" % path)\n========================== [FAILED] Took 0.35 seconds =========================="

JamesCCorralTech, 2022-09-21:
I encounter this issue currently when trying to create a new project for a Particle Boron or Particle Argon. I have not looked at other particle devices.

manually setting the board to "particle_boron" in platformio.ini causes the same error upon build.

mrclark25, 2022-09-21:
This is caused by particle_boron.json missing the softdevice configuration.

closed 2022-09-30 by stale[bot]

camow7, 2022-10-25:
This is the error I'm getting when attempting to flash it.

```
KeyError: "Invalid board option 'build.softdevice.sd_fwid'":
  File "C:\Users\catch\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 188:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\catch\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\catch\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\catch\.platformio\platforms\nordicnrf52\builder\main.py", line 142:
    board.get("build.softdevice.sd_fwid"),
  File "C:\Users\catch\.platformio\penv\lib\site-packages\platformio\platform\board.py", line 47:
    raise KeyError("Invalid board option '%s'" % path)
```

JamesCCorralTech, 2022-10-25:
@camow7 That looks like exactly the error I was getting and is fixed by the changes in my PR.

mrclark25, 2022-10-25:
@camow7 I believe the change in my PR will fix exactly that error.