platformio/platformio-home #5622: PIO home goes blank when click on platform details

ajaybhargav, 2023-04-07:
This might be happening on just one platform, here are the steps to reproduce:

1. Open PIO Home
2. Go to Platforms
3. Search "Logicrom" in list of Embedded platform
4. Click on platform name for details
5. Page goes blank

This is somehow happening only to Logicrom platform.

I opened PIO home in browser to see error in console, here it what says:

----

The above error occurred in task ms
    created by anonymous
    created by takeLatest(LOAD_PLATFORM_DATA, )
    created by na
    created by na
Tasks cancelled due to error:
takeLatest(LOAD_PLATFORM_DATA, )

----

Installation details:

OS: Windows 11
PIO Core: v6.1.6
PIO Home: v3.4.3

maxgerhardt, 2023-04-07:
Yesm the same issue appears for me. 

[image]

Seems like the platform doesn't have `url` defined in its metadata.a

maxgerhardt, 2023-04-07:
Using `pio pkg install -g -p "https://github.com/waybyte/platform-logicrom.git"` on the CLI works without errors though and after that the platform is visible as normal. Weird. `url` seems also defined in the git repo. Maybe it's an issue with the uploaded PIO registry package for the platform lacking `url`? 

[image]

ajaybhargav, 2023-04-07:
Hi @maxgerhardt  thank you so much for validating. I could not find issue with platform.json file which I thought might be causing issue. Is it PIO Home issue? I am not able to see this issue with any other platform though.

> Maybe it's an issue with the uploaded PIO registry package for the platform lacking `url`?

I also think that might be possible, but only devs can validate. I will however update my platform.json as I am preparing for a new release.

ajaybhargav, 2023-04-10:
I took reference from https://github.com/platformio/platform-espressif32/blob/develop/platform.json to update this https://github.com/waybyte/platform-logicrom/blob/master/platform.json

but it issue is still there. There is no indication about which URL is it looking for? I removed URL which existed earlier i thought that was issue.

maxgerhardt, 2023-04-10:
I may have misread the error message a bit, "Cannot read properties of undefined (reading 'url')" probably means that the object itself was null, not the url. As in `platform_object.url`, the `platform_object` is `null`, not that `url` is not there.

I think this needs @ivankravets's intervention to see what's going on there.

ajaybhargav, 2023-04-11:
I understand, I misinterpreted it too. The javascript is all minified so Its hard to understand from where the error in thrown. I have also emailed @ivankravets hope he checks it. It seems some information is pulled from PlatformIO registery after the page is loaded as there is a delay before page goes blank. The same information is not loaded once platform is installed.

closed 2023-04-11 by ivankravets