platformio/platformio-home #5628: Could not initialize project

kakhanjee, 2023-04-09:
PIO Core Call Error: "The current working directory C:\\Users\\Dream\\OneDrive\\Documents\\PlatformIO\\Projects\\countlcd will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\Dream\\OneDrive\\Documents\\PlatformIO\\Projects\\countlcd\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Traceback (most recent call last):\r\n  File \"C:\\Users\\Dream\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 102, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"C:\\Users\\Dream\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1130, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"C:\\Users\\Dream\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1055, in main\r\n    rv = self.invoke(ctx)\r\n  File \"C:\\Users\\Dream\\.platformio\\penv\\lib\\site-packages\\platformio\\cli.py\", line 71, in invoke\r\n    return super().invoke(ctx)\r\n  File \"C:\\Users\\Dream\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1657, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\Dream\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1404, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"C:\\Users\\Dream\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 760, in invoke\r\n    return __callback(*args, **kwargs)\r\n  File \"C:\\Users\\Dream\\.platformio\\penv\\lib\\site-packages\\platformio\\project\\commands\\init.py\", line 75, in project_init_cmd\r\n    init_base_project(project_dir)\r\n  File \"C:\\Users\\Dream\\.platformio\\penv\\lib\\site-packages\\platformio\\project\\commands\\init.py\", line 162, in init_base_project\r\n    config.save()\r\n  File \"C:\\Users\\Dream\\.platformio\\penv\\lib\\site-packages\\platformio\\project\\config.py\", line 453, in save\r\n    with open(path or self.path, mode=\"w+\", encoding=\"utf8\") as fp:\r\nFileNotFoundError: [Errno 2] No such file or directory: 'C:\\\\Users\\\\Dream\\\\OneDrive\\\\Documents\\\\PlatformIO\\\\Projects\\\\countlcd\\\\platformio.ini'\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq/index.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

kakhanjee, 2023-04-09:
bla bla bla

yifanshu02, 2023-04-10:
i have 2 PC.one of these is same yours.other is OK............