platformio/platformio-home #6332: Could not initialize project

Ricky1966, 2023-09-15:
PIO Core Call Error: "The following files/directories have been created in C:\\Users\\Utente\\Arduino ESP32 Nano\\NFC-Spools\r\ninclude - Put project header files here\r\nlib - Put project specific (private) libraries here\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\nResolving arduino_nano_esp32 dependencies...\r\n\n\nUnknownBoard: Unknown board ID 'arduino_nano_esp32'"

Ricky1966, 2023-09-15:
[env:arduino_nano_esp32]
platform = https://github.com/platformio/platform-espressif32.git
board = arduino_nano_esp32
framework = arduino

closed 2023-09-15 by Ricky1966

sylvanoMTL, 2024-06-21:
Hi Ricky, 
Can you please provide more details, did you create the project through the project creation wizard, or did you configure it manually in the platformio.ini file?