platformio/platformio-home #6384: Could not initialize project

vitauser1, 2023-09-24:
Error message:

PIO Core Call Error: "The following files/directories have been created in C:\\Users\\phili\\Documents\\PlatformIO\\Projects\\test2\r\ninclude - Put project header files here\r\nlib - Put project specific (private) libraries here\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\nResolving lilygo-t-display-s3 dependencies...\r\n\n\nUnknownBoard: Unknown board ID 'lilygo-t-display-s3'"

Trying to use platformio with Liliygo T display board - can't open a project!

I googled and saw this issue :   "Board "LilyGo T-Display-S3" fails to initialize #1071 "

but I was unsure how to fix following this.  I tried running:   pio pkg update -g -p espressif32 in the IDE terminal.  But in terminal in the platform gave error:

PS C:\Users\phili\Documents\T-Display-S3-main\T-Display-S3-main> pio pkg update -g -p espressif32
pio : The term 'pio' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included, verify that the path is 
correct and try again.
At line:1 char:1
+ pio pkg update -g -p espressif32
+ ~~~
    + CategoryInfo          : ObjectNotFound: (pio:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

Any help most gratefully received!

vitauser1, 2023-09-24:
I managed to resolve this.  Realised needed to update platforms?  Found platformio, project tasks, default, dependencies, update.   That seemed to work.   Writing this to help others!

closed 2023-09-24 by vitauser1

zakkhoyt, 2023-11-03:
@vitauser1 Yes! Thank you VERY much for posting your solution here. I have a similar open issue which I can now close.