platformio/platformio-home #6571: Could not initialize project

dduft, 2023-10-26:
PIO Core Call Error: "The following files/directories have been created in /home/dduft/Projects/PlatformIO/ZylinderHeadTemp\ninclude - Put project header files here\nlib - Put project specific (private) libraries here\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\nResolving esp32-c3-devkitc-02 dependencies...\n\n\nUnknownBoard: Unknown board ID 'esp32-c3-devkitc-02'"

ksuprynowicz, 2023-11-14:
I had the same problem. JSON file for this board seems to be missing in Platform IO installation.

https://github.com/platformio/platform-espressif32/blob/master/boards/esp32-c3-devkitc-02.json

Adding it resolves the issue.

dmosberger, 2025-01-26:
esphome is still defaulting to using platformio/espressif32 v5.4.0, which is rather old.  To get support for esp32-c3, you need a newer version of that package.  The easiest way to ensure that is to add a line of the form `platform_version: 6.10` to your yaml config.  For example:

```yaml
esphome:
  name: test_project
esp32:
  board: esp32-c3-devkitc-02
  framework:
    type: arduino
    platform_version: 6.10.0
 : 
 etc...
 : 
```