platformio/platformio-home #6583: Could not initialize project

goodmanforweb, 2023-10-28:
PIO Core Call Error: "The following files/directories have been created in /Users/bigmanhu/Documents/PlatformIO/Projects/led\ninclude - Put project header files here\nlib - Put project specific (private) libraries here\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nInvalidEnvNameError: Invalid environment name 'STC89C516RD+'. The name can contain alphanumeric, underscore, and hyphen characters (a-z, 0-9, -, _)"

goodmanforweb, 2023-10-28:
vscode

goodmanforweb, 2023-10-29:
[image]

goodmanforweb, 2023-10-29:
I find how to resolve this bug。I found PlatformIO Core code package on my computer, commented out the two lines of code inside, and then it was created successfully.
[image]

linux-tools, 2025-05-06:
好巧不巧，从CSDN到Github😏