platformio/platformio-home #6647: Unknown board ID 'seeed_xiao_esp32s3'"

tfcroft4, 2023-11-08:
Setting up a new project I get this error message?
seeed_xiao_esp32s3 is listed as a board in the Select a Board drop down.
Any thoughts
Ted Finch
PIO Core Call Error: "The following files/directories have been created in E:\\Users\\Ted\\Documents\\PlatformIO\\ESP32Projects\\ESP32_Test_2\r\ninclude - Put project header files here\r\nlib - Put project specific (private) libraries here\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\nResolving seeed_xiao_esp32s3 dependencies...\r\n\n\nUnknownBoard: Unknown board ID 'seeed_xiao_esp32s3'"

MikeTheGent, 2023-12-16:
Without knowing what I'm doing, I am attempting to kludge this by creating a seeed_xiao_esp32s3.json from seeed_xiao_esp32c3.json and making some edits. I will attach this file in the next reply.

I also created directory ~/.platformio/packages/framework-arduinoespressif32/variants/XIAO_ESP32S3 and copied ../XIAO_ESP32C3/pins_arduino.h there. (This is a maxOS path).

MikeTheGent, 2023-12-16:
Kludged seeed_xiao_esp32s3.json attached.
[seeed_xiao_esp32s3.json](https://github.com/platformio/platformio-home/files/13691708/seeed_xiao_esp32s3.json)

I won't know if this works until my XIAO ESP32S3 arrives in the next day or three. I do at least get a clean compile.